MrChromebox/firmware #712: Need help with supposedly bricked SASUKE

TheVorkMan, 2025-01-09:
I successfully installed UEFI firmware, then I successfully installed Linux.

Previously, I detached battery because it was dead and my Chromebook won't power on with it (I assume because of how long it was lying on my shelf untouched).

Just in case, after installation of Linux, I connected the battery and, as expected, there was no response on power key press, as how it was earlier before UEFI firmware.
But on ChromeOS I was able to boot in to recovery mode with battery attached, when detached I could power it on.

But, when I tried to start it again without battery with power supply connected - it doesn't start and LED blinks red then blue, and repeats.
I tried connecting batter again, then trying to press ESC + Reload + Power, but it didn't work.

MrChromebox, 2025-01-09:
Recovery mode doesn't exist with the UEFI firmware.

I'd replace the battery then see what's what.

TheVorkMan, 2025-01-09:
I wish I was able to. 

There is batteries from XE340XDA, is it compatible with XE345XDA? It has the same battery model and the same capacity.

MrChromebox, 2025-01-09:
I have absolutely no idea, completely unfamiliar with the device. I would open it up and check the battery model and search for that / cross compatibility.

TheVorkMan, 2025-01-09:
Yo, I have found out that my other very similiar Samsung laptop (ARM-based by the way, but sadly Windows) has the same battery. I will try to swap it.

TheVorkMan, 2025-01-09:
IT WORKED! Thx for help.