MrChromebox/firmware #714: Won’t boot without internal display connected on Samsung Chromebook 2 11 - XE500C12 Chromebook Winky after firmware update

henny-0, 2025-01-11:
Just updated to the latest MrChromebox firmware (MrChromebox-2408.1) for my Chromebook and and realized through trial that the device won’t boot without the internal display connected. Prior to this I was on firmware version 4.20.1 and could boot with just the HDMI connected to a monitor (not sure about headless booting). This would also show the Bios screen.

Now it just powers up (shown by the led indicator) if the internal display isn’t connect, but there is no indication of anything else happening. Pressing the power button once even after 5 minutes turns off the led. HDMI shows no indication of receiving a signal through all this period.

If I power down and attach the internal display, everything works. HDMI output operates like it did with the screen attached before, only after the bios screen.

robensam, 2025-02-10:
Just to confirm...Same for Dell chromebook 11.  Wont boot without internal display connected.

MrChromebox, 2025-02-10:
@robensam 'Dell Chromebook 11' could be any one of 5 or 6 devices.

robensam, 2025-02-10:
> [@robensam](https://github.com/robensam) 'Dell Chromebook 11' could be any one of 5 or 6 devices.

Model: P22T
Dell Chromebook 11 (3120) | CANDY

closed 2026-11-04 by MrChromebox