MrChromebox/firmware #715: Dell 3100 2 in 1, Grabbiter PowerUP on lid opening

A4EVA, 2025-01-12:
Hello, I successfully turned my Dell Chromebook 3100 into a fully working linux laptop thanks to Mr. Chromebox Script. Everything is perfect but I noticed that each time I lift the screen the device turns on If I remember correctly this was happening also with chrome os. I would like to disable this feature and only use the power button on the side of the unit. It is possible to do that by changing some options in the coreboot (even if a coreboot rebuild is needed) ?

MrChromebox, 2025-01-12:
this is "working as designed" -- all Chromebooks work this way.

It should be possible to change this by modifying the board's EC WAKE flags but I've not tested this myself.

I'd also like to make this a configurable option in the setup menu down the road

A4EVA, 2025-01-12:
I can try to do that and report to you if you explain me a bit what I need to change into the coreboot sources, I'm really new to the coreboot project :-)

closed 2026-11-04 by MrChromebox