MrChromebox/firmware #716: Asus Chromebook c204 problem working audio and touchpad.

BGT00, 2025-01-12:
Hello, I changed from Chrome OS to Windows, The problem occurred after installing the system that the touchpad and sound did not work. I started installing the drivers and with Intel Smart Sound OLED I have an exclamation mark and the code 10 Status devices power falure and the touchpad does not require any driver and is not available in the mouse tab

BGT00, 2025-01-12:
ps: the touchpad started working, user coolstar made an installer that triggered the touchpad
just what to do with the sound?

MrChromebox, 2025-01-12:
Windows/OS support is handled by the chrultrabook project - please read the docs there and post on the chrultrabook forum if you need help with OS drivers

closed 2025-01-12 by MrChromebox