MrChromebox/firmware #717: TANIKS - UEFI renders eMMC invisible

matyb, 2025-01-18:
> i think this was recently reintroduced somehow... everything was so straightforward and easy (given everything you're abstracting!) I'm confident I followed the chrultrabook guide to a T and used your most recent `firmware-utility.sh` and the eMMC storage is invisible to the uefi shell and any os i boot from usb into.
> 
> to be sure i restored to stock chrome and repeated, being super careful to give everything the time it needs, and the result is the same - eMMC is not recognized by my system.
> 
> i zipped up the [logs here](https://drive.google.com/file/d/1DCAPDmCmNcXrtSlW0D5dKPlVmDrU2LGY/view?usp=sharing) and actually made that link available to anyone. is there a post install step for the models that do not have nvme that isn't described here or anywhere in the help forums or reddit? lol i feel like i've looked everywhere and everyone's solution is to just install nvme 

 _Originally posted by @matyb in [#470](https://github.com/MrChromebox/firmware/issues/470#issuecomment-2599731435)_

i dont have permission to reopen, but i think a regression may have been introduced for taniks wrt emmc

MrChromebox, 2025-01-18:
The stock firmware selectively enables either eMMC or NVMe based on the value programmed in fw_config (small eeprom). I removed this check and enable both devices. Anything beyond that is likely an issue with the edk2 eMMC driver.

looking at your cbmem log, the eMMC controller on PCIe RP9 is enabled, so there's no issue on the coreboot side of things. I assume the eMMC is present once linux is booted from USB?

matyb, 2025-01-18:
nope, i don't see anything except the usb key i booted from in ubuntu 23.04, 23.10, 24.10, arch linux, and fedora 41.

i kinda assumed dmesg output would have been in the debug logs, but i'm unsure, it looks as though the OS sees nothing there at all.

matyb, 2025-01-18:
if coreboot checks out then - it sounds like i need to snoop around changes to https://github.com/tianocore/edk2 then?

MrChromebox, 2025-01-18:
my firmware completely ignores the storage setting in CBI and enables both, which should be all that's needed.

can you send a cbmem from the stock firmware? can grab from ChromeOS or Linux via RWL

matyb, 2025-01-18:
wish i could read this lol, had a lil trouble doing it in chromeos, but reinstalling the rw_legacy fw and booting into ubuntu i think it'll work 🤞 

https://drive.google.com/file/d/1S0ICZBfFlooII7pewb7fd4eCaBJY4t6Q/view?usp=sharing

thanks again for all of your help!!!

MrChromebox, 2025-01-18:
> had a lil trouble doing it in chromeos (permission denied even with execute flag on and exec'ing as root 🤔 ), but the linux via rwl route did the trick first try
> 
> https://drive.google.com/file/d/1zs2EowcPxMo65E-1VJN9n1MXhkbzvTes/view?usp=sharing
> 
> i do see all the mmcblk partitions under /dev there so 🙏
> 
> thanks again for all of your help!!!

seems you uploaded the cbmem binary and not the log

matyb, 2025-01-18:
lol i did originally! but immediately edited it... so sorry to have wasted your time, the link should point to the log file now

matyb, 2025-01-18:
i clicked it and got this from a guest window, im guessing somehow it's sending you something different?
```

coreboot-v1.9308_26_0.0.22-26476-g9dd4100506 Tue Oct  4 08:08:51 UTC 2022 bootblock starting (log level: 8)...
CPU: 12th Gen Intel(R) Core(TM) i3-1215U
CPU: ID 906a4, Alderlake R0 Platform, ucode: 00000423
CPU: AES supported, TXT NOT supported, VT supported
Cache: Level 3: Associativity = 10 Partitions = 1 Line Size = 64 Sets = 16384
Cache size = 10 MiB
MCH: device id 4609 (rev 04) is Alderlake-P
PCH: device id 5182 (rev 01) is Raptorlake-P SKU
IGD: device id 46b3 (rev 0c) is Alderlake P GT2
VBOOT: Loading verstage.
FMAP: Found "FLASH" version 1.1 at 0x1804000.
FMAP: base = 0x0 size = 0x2000000 #areas = 37
FMAP: area COREBOOT found @ 1875000 (7909376 bytes)
CBFS: mcache @0xfef85600 built for 72 files, used 0xfd8 of 0x2000 bytes
CBFS: Found 'fallback/verstage' @0x189540 size 0x17188 in mcache @0xfef8597c

coreboot-v1.9308_26_0.0.22-26476-g9dd4100506 Tue Oct  4 08:08:51 UTC 2022 verstage starting (log level: 8)...
Probing TPM I2C: I2C bus 1 version 0x3230302a
DW I2C bus 1 at 0xfe022000 (400 KHz)
done! DID_VID 0x00281ae0
TPM ready after 0 ms
cr50 TPM 2.0 (i2c 1:0x50 id 0x28)
Firmware version: B2-C:0 RO_A:0.0.13/b9f09fb5 RW_B:0.5.271/cr50_v4.1
Current CR50_BOARD_CFG = 0x80000001, matches desired = 0x00000001
tlcl_send_startup: Startup return code is 0
TPM: setup succeeded
src/security/tpm/tss/tcg-2.0/tss.c:231 index 0x1007 return code 0
src/security/tpm/tss/tcg-2.0/tss.c:231 index 0x1008 return code 0
Chrome EC: UHEPI supported
Reading cr50 boot mode
Cr50 says boot_mode is VERIFIED_RW(0x00).
Phase 1
FMAP: area GBB found @ 1805000 (458752 bytes)
MMAP window: SPI flash base=0x1000000, Host base=0xff000000, Size=0x1000000
MMAP window: SPI flash base=0x500000, Host base=0xf9500000, Size=0xb00000
VB2:vb2_check_recovery() Recovery reason from previous boot: 0x0 / 0x0
Phase 2
Phase 3
FMAP: area GBB found @ 1805000 (458752 bytes)
FMAP: area VBLOCK_A found @ 500000 (65536 bytes)
FMAP: area VBLOCK_A found @ 500000 (65536 bytes)
VB2:vb2_verify_keyblock() Checking keyblock signature...
VB2:vb2_verify_data() HW crypto for hash_alg 3 not supported, using SW
VB2:vb2_verify_digest() HW RSA for sig_alg 4 not supported, using SW
VB2:vb2_rsa_verify_digest() HW modexp for sig_alg 4 not supported, using SW
FMAP: area VBLOCK_A found @ 500000 (65536 bytes)
FMAP: area VBLOCK_A found @ 500000 (65536 bytes)
VB2:vb2_verify_fw_preamble() Verifying preamble.
VB2:vb2_verify_data() Using HW crypto engine for hash_alg 2
VB2:vb2_verify_digest() HW RSA for sig_alg 4 not supported, using SW
VB2:vb2_rsa_verify_digest() HW modexp for sig_alg 4 not supported, using SW
Phase 4
FMAP: area FW_MAIN_A found @ 510000 (5242816 bytes)
VB2:vb2api_init_hash() Using HW crypto engine for hash_alg 2
VB2:vb2_verify_digest() HW RSA for sig_alg 4 not supported, using SW
VB2:vb2_rsa_verify_digest() HW modexp for sig_alg 4 not supported, using SW
Saving vboot hash.
Saving nvdata
TPM: Extending digest for `VBOOT: boot mode` into PCR 0
tlcl_extend: response is 0
TPM: Digest of `VBOOT: boot mode` to PCR 0 measured
TPM: Extending digest for `VBOOT: GBB HWID` into PCR 1
tlcl_extend: response is 0
TPM: Digest of `VBOOT: GBB HWID` to PCR 1 measured
tlcl_lock_nv_write: response is 0
tlcl_lock_nv_write: response is 0
Slot A is selected
[... 2286 lines cut ...]
   1 edk2            altfw/edk2      edk2 UEFI bootloader
get_bootloader_menu: num_bootloaders: 1
get_bootloader_menu: Bootloader: filename=altfw/edk2, name=edk2, desc=edk2 UEFI bootloader, seqnum=1
get_bootloader_menu: Cached with 4 item(s)
ui_loop_impl: <Select alternate bootloader> menu item <edk2>
ui_display: screen=0x340, locale=0, selected_item=1, disabled_item_mask=0x0, hidden_item_mask=0x0, timer_disabled=1, current_page=0, error=0x0
get_bootloader_menu: num_bootloaders: 1
get_bootloader_menu: Bootloader: filename=altfw/edk2, name=edk2, desc=edk2 UEFI bootloader, seqnum=1
get_bootloader_menu: Cached with 4 item(s)
ui_menu_select: Menu item <edk2> run action
developer_boot_altfw_impl: Try booting from bootloader #1
Running bootloader 'edk2: edk2 UEFI bootloader'
Loading altfw/edk2 into RAM
CODE/DATA: dst=0x800000 dst_len=16777216 src=0x312375c8 src_len=824111 compression=1
TPM: secdata_kernel unchanged
TPM: secdata_firmware unchanged
nvdata unchanged
tpm_get_response: command 0x121, return code 0x0
TPM: secdata_kernel locked
Shutting down all USB controllers.
Removed USB disk 2.
Finalizing coreboot
Exiting depthcharge with code 8 at timestamp: 6471971
Starting altfw/edk2 at 0x8016b3...
```

MrChromebox, 2025-01-18:
hmm, for some reason coreboot is not scanning/finding the PCIe device attached to the root port - will have to see what's changed there recently

matyb, 2025-01-18:
hmu if you want a hand testing. im not terribly proficient in C or Bash, but am probably capable of figuring out the toolchain to build/compile/deploy.

paypalling ya now for your efforts. you really are a saint!

matyb, 2025-01-18:
that really helps! im wayyyyyy out of my element, but i wonder if enabling both 100% of the time didn't clash with this change:
https://chromium.googlesource.com/chromiumos/third_party/coreboot/+/92c8e51ee3867d4b0c1adebee2463071dbd3f8ec%5E%21/

from 6/30/2024
```
...
3. Disable eMMC controller incase STORAGE_UFS or STORAGE_NVME fw_config
is enabled.
...
```

MrChromebox, 2025-01-18:
TANIKS doesn't have any such enablement, just the FW_CONFIG checks in devicetree that are now removed.

Also, Alderlake-P/U boards like TANIKS don't support eMMC natively (unlike Alderlake-N boards like the one in the commit you linked to), they're using a PCIe to eMMC bridge chip, and I'm guessing that's where the issue is. If you swapped it out for a NVMe drive I'm guessing everything would be fine

matyb, 2025-01-18:
likewise, that seems to be what everyone else is doing to get around this limitation 😿 

i can PR the change to the supported devices if that's the direction the project would like. of if you think it's fixable with sufficient interest i'd be happy to dig in or hold out.

matyb, 2025-01-18:
Looking at how the page renders I believe this would make it clear, though it's kinda hacky lol

https://github.com/MrChromebox/website/pull/30

MrChromebox, 2025-01-18:
would prefer to fix the issue lol, I'm pretty sure it worked with a previous version of my firmware

MrChromebox, 2025-01-21:
@matyb do you have a SuzyQ cable? if so, I'll have you flash some earlier versions and see if we can locate the regression

matyb, 2025-01-21:
Nope I just dc'd the battery, but I'll be happy to score one if it's helpful. I'll order one now 🤘

matyb, 2025-01-21:
when the cable arrives - is the plan to bisect from MrChromebox-4.20.1 (and once confirming booting into a usb linux live env can find the /dev/mmcblk partitions) to the point where it is no longer visible?

is there a cache of released rom binaries to save on the [compilation steps](https://docs.mrchromebox.tech/docs/support/compiling.html)? i saw the release tags to build from, but no formal releases in github to just dl and flash.

MrChromebox, 2025-01-21:
I can provide the binaries to flash - building older versions can be problematic due to toolchain issues.

first we care about coreboot initializing the eMMC - that will allow Linux to see it regardless of whether edk2 does or not.
next we'll look at edk2 once the coreboot side is resolved

matyb, 2025-01-23:
USPS is saying 1/27, but I'm itching to move this forward 😈 

Let me know which image to flash first and I'll get moving the second that bad boy arrives (and I'm off work)

matyb, 2025-01-27:
@MrChromebox I've got a suzyqable in hand and am ready to roll, let me know what you think first steps are 🙏

MrChromebox, 2025-01-27:
@matyb let's flash these and see where the breakage is.

4.20.1
[coreboot_edk2-taniks-mrchromebox_20230721.zip](https://github.com/user-attachments/files/18564323/coreboot_edk2-taniks-mrchromebox_20230721.zip)

4.21.1
[coreboot_edk2-taniks-mrchromebox_20231029.zip](https://github.com/user-attachments/files/18564326/coreboot_edk2-taniks-mrchromebox_20231029.zip)

4.22.2
[coreboot_edk2-taniks-mrchromebox_20240223.zip](https://github.com/user-attachments/files/18564336/coreboot_edk2-taniks-mrchromebox_20240223.zip)

2405
[coreboot_edk2-taniks-mrchromebox_20240615.zip](https://github.com/user-attachments/files/18564343/coreboot_edk2-taniks-mrchromebox_20240615.zip)

these can all be flashed manually with flashrom per the docs.
please report for each version if eMMC is shown in edk2 boot menu and visible under booted OS (if booted from USB)

matyb, 2025-01-27:
RW_LEGACY from most recent release (as of 1/14/2024) EDK2 🔴 EMMC in OS ✅ 

4.20.1
[coreboot_edk2-taniks-mrchromebox_20230721.zip](https://github.com/user-attachments/files/18564323/coreboot_edk2-taniks-mrchromebox_20230721.zip) EDK2 🔴 EMMC in OS 🔴
4.21.1
[coreboot_edk2-taniks-mrchromebox_20231029.zip](https://github.com/user-attachments/files/18564326/coreboot_edk2-taniks-mrchromebox_20231029.zip) EDK2 🔴 EMMC in OS 🔴 
4.22.2
[coreboot_edk2-taniks-mrchromebox_20240223.zip](https://github.com/user-attachments/files/18564336/coreboot_edk2-taniks-mrchromebox_20240223.zip) EDK2 🔴 EMMC in OS 🔴
2405
[coreboot_edk2-taniks-mrchromebox_20240615.zip](https://github.com/user-attachments/files/18564343/coreboot_edk2-taniks-mrchromebox_20240615.zip) EDK2 🔴 EMMC in OS 🔴

With the exception of the RW_LEGACY flash I was starting from (which was fresh off a total restore using a chromeos recovery usb) I was unable to see the EMMC when booting into Ubuntu 24.10. Nothing showed the EMMC drive in EDK2's menu.

From https://github.com/MrChromebox/firmware/issues/470 the comments hint at the EMMC storage being visible once booted into linux, but I did not see this.

I was able to see all the volumes on the left hand side of the gnome taskbar, but also see them listed under `/dev/mmc*` and just to be sure the installer wasn't initializing something I walked through the installer until I selected a disk, and at that point only my usb key was visible -  except from the RW_LEGACY flash I started from (it showed my emmc as a device I could wipe partitions and install to).

I'm guessing the EMMC storage never was actually seen, I have been unable to find any evidence (aside from the comment in #470) I haven't seen any evidence it's initialized from UEFI. So maybe this isn't a regression, but a misunderstanding and support would be entirely new.

For flashing, I followed the steps provided on the [official docs](https://docs.mrchromebox.tech/docs/firmware/manual-flashing.html) and everything went to plan, my hwid looked exactly like stock chromeos developer mode reported it, flashing ended with VERIFIED, etc. each reboot i just let it do its thing for like 60s to start booting and didn't mess with power/battery/etc. so I'm pretty sure the flashing was spot on...

matyb, 2025-01-27:
im game to get any logs, do any remoting, can pair, test debug builds, test special one off builds, whatever you're up for. i really appreciate your help even getting this far with a solution. if i end up needing an nvme at the end of all this, that's ok too, i know the pcie -> emmc chip is super niche and may not yield any additional device support.

there's some specs on the specific chip but im unsure how to find any additional information about it [here](https://www.reddit.com/r/chromeos/comments/155ep48/lenovo_gaming_16_inch_taniks_is_emmc_storage/)

all i can find is its a `Genesys Logic, Inc GL9763E eMMC Controller (rev 02)`

matyb, 2025-01-28:
I did capture the debug logs from 4.20.1 since before I started flashing I was pretty confident it "worked" I've made them available [here](https://drive.google.com/file/d/15RVFjREWdnd9wDlA-6katVwMDV--AUEa/view?usp=sharing)

matyb, 2025-01-28:
> here's the stock cbmem.log with a totally new link and a different google account 🤞 with the stock or rwl fw where the emmc is visible to OS's: https://drive.google.com/file/d/1j-F0cI9PhIQSrlOGNtbeOwPB5M8weqyk/view?usp=sharing
> 
> and the one i uploaded originally along with the rest of the logs, but to save you the search and scan this is the uefi cbmem.log where the emmc isn't visible to the os: https://drive.google.com/file/d/1JQtMBZOfDQRXGSPDP8xwOilVtkD9DuLD/view?usp=sharing

I did capture the debug logs while i had 4.20.1 flashed since before I started flashing I was pretty confident it "worked" I've made them available [here](https://drive.google.com/file/d/15RVFjREWdnd9wDlA-6katVwMDV--AUEa/view?usp=sharing)

MrChromebox, 2025-02-03:
someone else was supposed to send me one for testing and never did, so if you're not using it I'll gladly take it and see if I can reproduce the issue with a similar ADL device

JDIacobbo, 2025-12-06:
> should be fixed in 2509.4 release, can someone confirm?

So I tried to test this by replacing the nvme drive I was trying to use to test #724 with the emmc drive the device came with and when I did I was unable to get the devices keyboard to work. Swapping back to the nvme drive and the keyboard work. The keyboard also works when there is no drive installed. I plugged in an external keyboard and that worked as well although I was unable to successfully load the Ubuntu live session (it just kept spinning). I have no idea if this is related to this issue and I'm wondering if someone else could test this.

*Edit*

I tried to boot gparted to check if I could at least see the drive and this is what I was met with.

[image]

MrChromebox, 2025-12-07:
> I'd be happy to help out were I can. Do you need the eMMC drive or the whole device with drive?

what device do you have? I thought I'd already fixed the eMMC drive, as it works on other devices using the same one