MrChromebox/firmware #719: How to Unlock DVMT to 64MB

MizhiDR, 2025-01-24:
My device is a lenovo yoga c630 chromebook, the CPU is i7-8550U, it has a 4K screen, and I recently installed Macos, but I encountered the problem of being unable to drive UHD620. I would like to ask how to unlock DVMT

MrChromebox, 2025-01-24:
DVMT is not adjustable/unlockable. If you want to change the amount, you have to edit the coreboot source code and rebuild.

MizhiDR, 2025-01-24:
> DVMT 不可调整/解锁。如果要更改数量，则必须编辑 coreboot 源代码并重建。

I need to change the DVMT value. How can I obtain the coreboot source code?

MrChromebox, 2025-01-24:
> I need to change the DVMT value. How can I obtain the coreboot source code?

looking at the source, 64MB is the largest amount fr DVMT. I can set it higher and build for you, but you'd need to be able to unbrick if something went wrong.

I can also adjust the Aperture size (pre-allocated amount): 128/256/512 mb

MizhiDR, 2025-01-24:
> > 我需要更改 DVMT 值。如何获取 coreboot 源代码？
> 
> 从源代码来看，64MB 是 DVMT 的最大容量。我可以将其设置得更高并为您构建，但如果出现问题，您需要能够解开。
> 
> 我还可以调整光圈大小（预分配量）：128/256/512 mb

I need DVMT value to be 128MB .Also, are you considering making a menu to try and have options?

MrChromebox, 2025-01-24:
> I need DVMT value to be 128MB .

there is no documented value for 128MB:
`0x00:0 MB, 0x01:32 MB, 0x02:64 MB`

I can try setting it to 3 or 4, but can't promise it will work

> Also, are you considering making a menu to try and have options?

in progress. hopefully for next release, maybe one after that

MizhiDR, 2025-01-24:
> > 我需要 DVMT 值为 128MB。
> 
> 没有记录 128MB 的值： `0x00:0 MB, 0x01:32 MB, 0x02:64 MB`
> 
> 我可以尝试将其设置为 3 或 4，但不能保证它会起作用
> 
> > 另外，您是否考虑制作菜单来尝试提供选择？
> 
> 正在进行中。希望在下一个版本中，或许在那之后的一个版本中

I thought about it and set it to 64MB.

MrChromebox, 2025-01-24:
> I thought about it and set it to 64MB.

64mb is the default

MizhiDR, 2025-01-24:
> > 我想了想，还是设置为64MB吧。
> 
> 64mb 是默认值

Sorry, I didn't know the default value is 64MB. I can try 128MB. I would like to ask you to add an optional menu. Even if 128MB does not take effect, I can change it back. Then I won't have to trouble you again.

MrChromebox, 2025-01-24:
I don't have the time to implement everything needed for the menu at this time. I can provide a binary with the value fixed to 3 or 4, which should be 96/128mb. I'll test here on a KBL device first

MrChromebox, 2025-01-24:
I will post the binary here for you to flash

MrChromebox, 2025-01-24:
> Thank you very much. Have you considered unlocking the power wall? I know this is a low-power processor.

adjusting power limits is one of the things which I plan to add to the option menu, once implemented

MizhiDR, 2025-01-24:
> > 非常感谢。你考虑过解锁电源墙吗？我知道这是一个低功耗处理器。
> 
> 调整功率限制是我计划在实施后添加到选项菜单中的功能之一
Understood, thank you for your efforts

MizhiDR, 2025-01-25:
> > Thank you very much. Have you considered unlocking the power wall? I know this is a low-power processor.
> 
> adjusting power limits is one of the things which I plan to add to the option menu, once implemented

One thing reminds me of whether this firmware supports UEFI boot. I am worried that the IGPU is not in UEFI boot mode.

MrChromebox, 2025-01-25:
this firmware only supports UEFI boot. The firmware loads the UEFI GOP driver as part of the boot process.

I'm not seeing a good way to verify the DVMT and Aperture sizes from the OS, so having trouble telling if my test builds are actually changing anything

MizhiDR, 2025-01-25:
> 此固件仅支持 UEFI 启动。固件在启动过程中加载 UEFI GOP 驱动程序。
> 
> 我没有找到一个很好的方法来验证操作系统的 DVMT 和 Aperture 大小，因此无法判断我的测试版本是否真的改变了任何东西

Could you please let me test it?

MrChromebox, 2025-01-26:
[coreboot_edk2-nami-mrchromebox_20250125.zip](https://github.com/user-attachments/files/18548635/coreboot_edk2-nami-mrchromebox_20250125.zip)

This is MrChromebox-2408 release + DVMT = 128MB, Aperture = 512MB.

flash this manually:
`sudo flashrom -p internal --fmap -i COREBOOT -w coreboot_edk2-nami-mrchromebox_20250125.rom -N`

use flashrom donwloaded by my script (saved to /tmp/flashrom) if needed

zirblazer, 2025-01-28:
> I'm not seeing a good way to verify the DVMT and Aperture sizes from the OS, so having trouble telling if my test builds are actually changing anything

Aperture Size is easy. In Linux you can see the IGP Aperture Size by using...
lspci -vvv -s 00:02.0 | grep "Region 2"
...which should return...
Region 2: Memory at c8000000 (64-bit, prefetchable) [size=512M]
size= is of course the value you want. Note that I tested this on desktop Haswell, not sure if it still the same.

MrChromebox, 2025-01-28:
then the aperture size change works up to 512mb

zirblazer, 2025-01-28:
> then the aperture size change works up to 512mb

I had some notes regarding IGP Aperture Size because it was reelevant for Intel GVT-g for GPU Virtualization, which limited the amount of VMs you could share IGP with based on aperture size. Haswell supports up to 512 MiB, Broadwell up to 1 GiB. Skylake apparently further extended it to 2 GiB, as you can see in this random manual I pulled: https://estore.lex.com.tw/uploads/product/2i610hw/2i610hw-manual.pdf

MizhiDR, 2025-02-08:
> > 然后光圈大小变化可达 512mb
> 
> 我有一些关于 IGP 孔径大小的注释，因为它与用于 GPU 虚拟化的 Intel GVT-g 相关，它根据孔径大小限制了可以共享 IGP 的虚拟机数量。Haswell 支持高达 512 MiB，Broadwell 支持高达 1 GiB。Skylake 显然将其进一步扩展到 2 GiB，正如您在我随机提取的手册中看到的那样：https://estore.lex.com.tw/uploads/product/2i610hw/2i610hw-manual.pdf
> 
> 我还记得，一些手册中有一些警告，称 2 GiB 仅在启用 4G 以上 MMIO 的情况下才有效。不知道这如何映射到 Coreboot。

Thank you for your help. I have enabled core graphics acceleration on macos10.14, but the acceleration cannot be enabled on macos11, 12, and 13. I don’t understand whether this is related to coreboot? Also, is the Intel Management Engine enabled?

MrChromebox, 2025-02-08:
I can't help with any MacOS related issues, I know nothing about it or Hackintoshing. There are other forums better suited to getting help with that.

MohammedAtique, 2025-05-31:
ChromeOS Device Firmware Utility Script [2025-05-21] 
 (c) Mr Chromebox <mrchromebox@gmail.com> 
*********************************************************
**     Device: Asus Chromebox 3 / CN65
** Board Name: TEEMO
**   Platform: Intel KabyLake
**    Fw Type: Full ROM / UEFI
**     Fw Ver: MrChromebox-2503.0 (04/27/2025)
**      Fw WP: Disabled
*********************************************************
** [WP] 1) Install/Update UEFI (Full ROM) Firmware 
** [WP] 2) Restore Stock Firmware 
**      C) Clear UEFI NVRAM 
*********************************************************
Select a numeric menu option or
R to reboot  P to poweroff  Q to quit 

Hey I wanna change aperture size to 2gb so is there any script or something to do that

MrChromebox, 2025-05-31:
@MohammedAtique that's not an available option for the Aperture Size, the available options are listed under the UEFI Platform Settings Menu

MohammedAtique, 2025-06-01:
> [@MohammedAtique](https://github.com/MohammedAtique) that's not an available option for the Aperture Size, the available options are listed under the UEFI Platform Settings Menu

But from UEFI platform there is no option to change

Here is something I found for reference: https://github.com/intel/gvt-linux/issues/131#issuecomment-815362316

I tried but didn’t get the varoffset value

MrChromebox, 2025-06-02:
> But from UEFI platform max value is 512mb only

that is the max supported for your device. If values > 512MB worked, I would have added them.