MrChromebox/firmware #721: Acer Chromebook R11 Tablet mode

Myrkie, 2025-01-27:
im using Onboard as an onscreen keyboard, Onboard supports Convertible devices that use SW_TABLET_MODE, is this supported by anything anywhere? libinput, evtest and acpid don't have any related events.

MrChromebox, 2025-01-27:
CYAN is I think the first convertible Chromebook, and the EC/sw mechanism for switching to tablet mode is a bit different than newer models. I have no idea if it's supported by mainline linux, or if it's ChromeOS specific.

I have one here I can pull out and try to see what's going on. What distro/kernel are you using?

Myrkie, 2025-01-29:
hey to follow up, anything else I should provide?

MrChromebox, 2025-01-29:
nothing else, I just haven't had time to look into the issue

Myrkie, 2025-01-29:
Alright, cheers!

MrChromebox, 2025-01-29:
If you're good manually flashing under Linux, I can throw a test build up here with the EC Tablet model controls enabled/exposed, and see if that helps

Myrkie, 2025-01-29:
that would be fine I still got my live CD ready with a more permissive kernel. 👍

mysfitt, 2025-01-30:
I believe I'm experiencing the same issue on an Acer CP311 (LAVA rev3). The Tablet mode switch does not emit any events detectable at the OS level (libinput, evtest, or acpi_listen), but flipping the screen does disable the keyboard, so it would seem that it's working at the EC level. I can confirm that with the ectool console:

> [92692.210877 KB disable_scanning_mask changed: 0x00000004]
> [92708.334555 KB disable_scanning_mask changed: 0x00000000]

Coreboot is the latest version for my machine at this time: MrChromebox-2408.1. The OS is  Arch Linux with the latest kernel as of this writing:  6.12.10-arch1-1. I also tried the LTS at version 6.6.72-1 with the same results. A Ubuntu Live distro I had on a USB key gave the same results as well. Note that I'm having a similar issue with the Lid switch on the same machine, but I'll create another bug report for that so as not to thread jack.

MrChromebox, 2025-01-31:
Had a chance to test CYAN today on Ubuntu 24.04.1, and can confirm it doesn't generate any tablet events.

I added support for `EC_ENABLE_TBMC_DEVICE` like newer devices have, and while that enables rotation, the device seems permanently stuck in tablet mode from the OS perspective, so the built-in keyboard and touchpad are disabled. This is obviously not usable so will require some further investigation.

Myrkie, 2025-02-03:
> and while that enables rotation
Interesting, I noticed with my testing that screen rotation was just completly busted and random and went with manual rotation.

> This is obviously not usable so will require some further investigation.
Thanks for looking into it in the first place. hope to see it fixed in the future though.

closed 2026-11-04 by MrChromebox