MrChromebox/firmware #723: ACER CP311 (LAVA) - Lid Switch does not emit events detectable by OS

mysfitt, 2025-01-30:
As stated, the Lid switch on my Acer CP311 rev3 does not emit an ACPI event that is detectable by the OS. I'm running Arch Linux at kernel version 6.12.10-arch1-1. Firmware is MrChromebox-2408.1. The 6.6 LTS kernel gives the same results. I've tried `libinput debug-events`, `acpi_listen`, and putting systemd-logind into debug mode, and none of those show an event when the lid is opened or closed. I can see the events from the ectool console though: 

> [94380.693081 lid close]
> [94380.693718 KB disable_scanning_mask changed: 0x00000001]
> [94380.704440 SW 0x00]
> [94381.005853 KB disable_scanning_mask changed: 0x00000005]
> [94383.132130 lid open]
> [94383.132763 KB disable_scanning_mask changed: 0x00000004]
> [94383.140944 SW 0x01]
> [94383.609462 KB disable_scanning_mask changed: 0x00000000]

And even more interesting is that the state in the proc filesystem is getting updated:

> cat /proc/acpi/button/lid/LID0/state 
> state:      open

I close the lid while in a ssh session to the machine...

> cat /proc/acpi/button/lid/LID0/state 
> state:      closed

This is the only place that I've found in the OS that's recognizing the event. For reference, here's a list of libinput devices for the machine:

> libinput list-kernel-devices
> /dev/input/event0:	Lid Switch
> /dev/input/event1:	Sleep Button
> /dev/input/event2:	Power Button
> /dev/input/event3:	Video Bus
> /dev/input/event4:	AT Translated Set 2 keyboard
> /dev/input/event5:	Intel Virtual Buttons
> /dev/input/event6:	Tablet Mode Switch
> /dev/input/event7:	PC Speaker
> /dev/input/event8:	Elan Touchscreen
> /dev/input/event9:	Elan Touchpad
> /dev/input/event10:	WCOM50C1:00 2D1F:0012 Stylus
> /dev/input/event11:	WCOM50C1:00 2D1F:0012
> /dev/input/event12:	keyd virtual keyboard
> /dev/input/event13:	keyd virtual pointer
> /dev/input/event14:	avs_da7219 Headset Jack
> /dev/input/event15:	hdaudioB0D2 HDMI/DP,pcm=1
> /dev/input/event16:	hdaudioB0D2 HDMI/DP,pcm=2
> /dev/input/event17:	hdaudioB0D2 HDMI/DP,pcm=3

I don't know if it's relevant, but the machine does not have a hardware sleep button as far as I know. Note that the Tablet Mode Switch is also not working at the OS level, which I mentioned as a comment in Issue #721.

MrChromebox, 2025-01-30:
Issue #721 is for a completely different device

mysfitt, 2025-01-30:
@MrChromebox Ok, sorry about that. I thought it could be related. I will delete my comment there and update this issue to include the information related to the Tablet switch.

mysfitt, 2025-01-30:
The Tablet mode switch does not emit any events detectable at the OS level (libinput, evtest, or acpi_listen), but flipping the screen does disable the keyboard, so it would seem that it's working at the EC level. I can confirm that with the ectool console:

> [92692.210877 KB disable_scanning_mask changed: 0x00000004]
> [92708.334555 KB disable_scanning_mask changed: 0x00000000]

I installed the Onboard soft keyboard, which supports SW_TABLET_MODE and it did not detect the tablet switch. The kernel enumerates the device as /dev/input/event6.

MrChromebox, 2025-01-30:
there are essentially 2 different tablet mode drivers that can be used at the OS level - the ChromeOS ones, or the Intel ones. Which are you using?

mysfitt, 2025-01-30:
I wasn't aware that there were two. As far as I can tell, there isn't a special kernel driver loaded for it. Looking at the dmesg output, it looks like the tablet switch was picked up by the standard uinput module:

> sudo dmesg|grep -i tablet
> [    5.632690] input: Tablet Mode Switch as /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:27/PNP0C09:00/GOOG0004:00/GOOG0006:00/input/input7

How can I tell which driver is in use? or is there a user space component that I'm missing?

Myrkie, 2025-01-31:
> Issue [#721](https://github.com/MrChromebox/firmware/issues/721) is for a completely different device

interestingly enough I downloaded and installed [ectool](https://docs.chrultrabook.com/docs/installing/ectool.html) and saw these changes and the keyboard did in fact turn off as it should, though onboard wouldn't change to tablet mode. just seems like `xinput test-xi2` doesn't receive any events

What I did
# Installation
<details>
<summary>sudo wget https://files.tree123.org/utils/x86_64/gnu/ectool</summary>
--2025-01-31 03:42:37--  https://files.tree123.org/utils/x86_64/gnu/ectool
Resolving files.tree123.org (files.tree123.org)... **.**.***.*
Connecting to files.tree123.org (files.tree123.org)|**.**.***.*|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 307552 (300K) [application/octet-stream]
Saving to: ‘ectool’

ectool              100%[===================>] 300.34K  --.-KB/s    in 0.1s    

2025-01-31 03:42:38 (1.96 MB/s) - ‘ectool’ saved [307552/307552]
myrkur@MyrkieCyan:&#126;/Downloads$ sudo chmod +x ./ectool
</details>

# Screen flipped 180Degrees into tablet mode
<details>
<summary>myrkur@MyrkieCyan:~/Downloads$ sudo ./ectool console</summary>
[178873.096470 KB disable_scanning_mask changed: 0x00000004]
</details>

# Screen in laptop mode
<details>
<summary>myrkur@MyrkieCyan:~/Downloads$ sudo ./ectool console</summary>
[178873.096470 KB disable_scanning_mask changed: 0x00000004]
[178880.202952 KB disable_scanning_mask changed: 0x00000000]
</details>

mysfitt, 2025-01-31:
@Myrkie I just wanted to confirm that I followed the exact same procedure and got the same results. What I suspect is that there is a sensor that is not being picked up by the OS. Maybe the driver doesn't exist or one would need to be modified to support the specific hardware? Out of curiosity, does your lid sensor work? I ask because I read something in the [EC documentation on the Chromium website](https://chromium.googlesource.com/chromiumos/platform/ec/+/HEAD/docs/ec_terms.md) that I think was a clue. Some Chromebooks have something called a GMR (Giant Magnetoresistance) sensor that controls both the Lid open/close events AND the tablet mode. It's actually pretty interesting. It works by detecting the strength of the magnetic field so it can tell the difference between closed, open, and tablet mode. I have no way of knowing which machines use this sensor though, or if there's a Linux driver for it in the kernel. I'm sure @MrChromebox would know better than you or I what hardware is present in our respective machines. It's entirely possible that an OS driver isn't even needed if the EC or firmware are doing the translation to emit ACPI events that the OS can pick up.

Myrkie, 2025-01-31:
that would be my guess too as mychromebox said in #721 

>CYAN is I think the first convertible Chromebook, and the EC/sw mechanism for switching to tablet mode is a bit different than newer models

MrChromebox, 2025-01-31:
ok, so LAVA is an Apollolake device, and they all use the same tablet mode profile: they define `EC_ENABLE_TBMC_DEVICE` which exposes the ACPI device `GOOG0006` which is the tablet mode switch.

I've booted REEF here (another APL device) with Ubuntu 24.04.1 live, and it correctly detects the `GOOG0006` device per dmesg. It's also loading the Intel virtual button driver (intel_vbtn) which may be conflicting, so you can try blacklisting that module and see if that helps. The VBTN driver support was added for Windows users.

mysfitt, 2025-01-31:
@MrChromebox I am seeing the `GOOG0006` device in my dmesg, but it's just not functional for some reason. I think you're probably right that it's some kind of conflict. I blacklisted the intel_vbtn and rebooted and there was no change, so that's not it. But it got me thinking it could be something else. I noticed that it's loaded two LPC drivers and also throwing some related errors from the kernel:
> sudo dmesg | grep lpc
> [    5.484888] cros_ec_lpcs GOOG0004:00: Chrome EC device registered
> [    5.486838] cros_ec_lpcs GOOG0004:00: Transfer error 1/3: -95
> [    5.682540] lpc_ich 0000:00:1f.0: I/O space for ACPI uninitialized

Maybe I should disable one of those and see what happens? Seems likely the switches might be under that bus.

mysfitt, 2025-02-02:
I just booted up on a Ubuntu Live 24.04.1 live usb stick and sure enough, the tablet functionality worked out of the box. So I must have something off in my OS config. At any rate, that's my problem to sort out and not any fault of the firmware. However, the Lid switch still did not work. It doesn't emit an ACPI event at all. Is it possible that this is a DSDT table thing? @MrChromebox could you boot up that REEF machine and see if the lid switch is working on it?

mysfitt, 2025-02-02:
tldr;

- Verified that hardware switches for lid and tablet work at the EC level
- The kernel does not emit any ACPI events for lid or tablet switches
- Tested both switches using sway's `bindswitch` function to bind a simple log action to each switch. Neither triggered.
- the iio accelerometer does work and will perform screen rotation under gnome. This is a separate function from the tablet switch

The long version...

After a bit more investigation, I determined that it's not the tablet switch that's triggering, it was actually the iio accelerometer that fooled me into thinking that the tablet switch had triggered. So at the OS level, I'm still not able to use the Lid switch or the Tablet switch, which are the two hardware switches present on the platform. To be clear, both devices do show up in dmesg and in the kernel device tree, but neither actually work. Systemd-logind, which handles the lid switch on most Linux distros these days, does not initiate suspend on lid closure, which is the default behavior on Ubuntu and Arch (the two distros I've tested with). `systemctl suspend` works fine. The tablet switch is harder to test, but I did configure swaywm to trigger on the tablet switch and it did not seem to be triggering. I used `bindswitch tablet:toggle exec 'logger TABLET_TOGGLE'` to test. I did something similar with `bindswitch lid:toggle` and it also did not trigger on lid closure. I'm quite certain that there are no ACPI events being triggered by the switches as I monitored with acpi_listen while running acpid. This seems to distro independent and I've also tested with a few kernel versions with the same results. The hardware does seem to actually work since the  EC disables the keyboard in tablet mode and the EC console log shows the lid open and close events. The machine also boots itself from power off when I open the lid. Which is a bit annoying, but does indicate that the lid switch hardware is working. 

I'm glad to do any more testing that you'd recommend, but at this point I'm not sure what else I can try. The lid bug has been driving me nuts as I'm in the habit of closing a laptop and walking away when I'm not using it and that results in a dead battery in this case. Please let me know if there's any other information I can provide. If there's a way I can downgrade firmware to test if this is a regression issue, I'd be glad to give that a shot. I think the firmware script will only upgrade, unless there's an option I'm not aware of. I'm pretty familiar with flashrom and fwupd though, which should work fine if I had the rom file.

MrChromebox, 2025-02-02:
> The lid bug has been driving me nuts as I'm in the habit of closing a laptop and walking away when I'm not using it and that results in a dead battery in this case

closing the lid on reef puts the device into suspend, as evidenced by the screen turning off and the flashing power LED.

I have no idea why it wouldn't be working for LAVA, every Chromebook with ChromeEC (so all since 2013) use the exact same ACPI for the lid switch.

mysfitt, 2025-02-03:
Apparently, I am not the only one. I came across a recent post by a user on Coolstar's forum today that has the exact same Chromebook as me and is having the [exact same problem](https://forum.chrultrabook.com/t/2-in-1-lid-switch-not-working/4597).

mysfitt, 2025-02-09:
As an additional data point, I booted my lava chromebook up on FreeBSD 14.2 today and set the lid switch to trigger S3 suspend. It also did not work. -_- So it's not the OS it would seem. I tried to test on Windows 10, but couldn't get my live usb stick to work. It essentially just went into a boot loop and never got to a working desktop.

I'm pretty sure the flash storage is soldered to the board, or else I'd just throw a spare mSATA drive in and do an actual Windows install. I'm not sure if there's a PCIe or M2 slot on the board. The PCI device tree shows one, but it could be a dead head.

closed 2026-11-04 by MrChromebox