MrChromebox/firmware #724: TANIKS (Lenovo gaming Chromebook) troubles with NMVe storage devices

Pinero99, 2025-02-05:
Hi, I'm fighting to get a stable Linux distro running, I spent several days debugging trying to find what's happening and I realized my laptop had troubles with that kind of device.

I flashed the latest firmware version available and after that I installed a debian distro in a 1TB  NMVe drive, after that I had multiple troubles with random hard freeze forcing to shutdown the laptop. 
no errors on logs, I tried running a few more distro (Linux mint and Kubuntu), used another NMVe drive to discard any malfunction on the first one and got the same trouble.

But I realized that when I was running a  Linux mint live session using a regular USB  drive,  it ran smoothly and never got freezing or crashes and just for curiosity I tried a  USB 3.0 SSD enclosure.

 I check with both NMVe drives running a full installed distros running from the enclosure and still have the same problem 

I tried to run a live session (Linux mint installation trial) and unlike the regular USB flash drive, that live session crashed too.

So basically, that machine didn't tolerate NMVe devices and I don't know why.

MrChromebox, 2025-02-05:
I see. There's not a lot of tuning on the firmware side for NVMe. Have you checked if there is a firmware update for the NVMe drive itself?

Pinero99, 2025-02-07:
Well I checked for firmware updates for those NMVe drive I have but since these two drive are "low budget" devices the brands didn't offer any support or firmware updates. Right now I finished with a roll back to chromeOS using the OG 128gb eMMC drive (which was recognized by the hardware running the OG chrome firmware) now laptop it's running "smoothly".

I'm wondering if I will sell or pawn this laptop, it's a chame cause it's a good hardware on papers but gave me a lot of compatibility troubles with linux during my last mouth and a half and chromeOS isn't a good is to my needs.

JDIacobbo, 2025-05-04:
I am having this same exact issue and progressed through the exact same troubleshooting steps. I have a TANKIS that came with a emmc drive I replace with a second hand nvme drive acquired via eBay (I believe is a 256 gb sk hynix). Booting any distro of Linux off of the nvme drive results in a hard lockup. I also tried booting a separate nvme drive from a USB enclosure to rule out a bad nvme drive but also experienced hard locks. Running a live environment from a USB works with out issue. 

I'd love to help figure this out because as OP mentioned this is a decent bit of hardware if we can get a stable experience.