MrChromebox/firmware #725: "command not found" when trying to download the seabios bin file

ciriousjoker, 2025-02-12:
Downloading the script works fine, then I'm trying to set up RW_LEGACY.

After answering the warning and declining the default boot, I get this error:

> ./firmware.sh: line 97: -sLO: command not found
> ./firmware.sh: line 98: -sLO: command not found
> 
> RW_LEGACY download checksum fail; download corrupted, cannot flash

I guess the download command is somehow broken/not available on my system?

I'm using the Acer Chromebook Spin 13 on beta channel.

MrChromebox, 2025-02-12:
please provide the full and exact command you are using, and where you are running it

ciriousjoker, 2025-02-12:
`cd; curl -LO mrchromebox.tech/firmware-util.sh && sudo bash firmware-util.sh`

I ran it in the VT2, ie ctrl+alt+f2, then logged in, typed out the command, pressed enter.
Then your script ran and I selected option 1 (RW_LEGACY) and pressed enter.
Answered the warning with y, answered the question about defaulting to usb boot with n.

Then, the script said "downloading seabios.bin" (not exactly, but something really close to that) and then it failed with the error message from above.

MrChromebox, 2025-02-12:
> Then, the script said "downloading seabios.bin" (not exactly, but something really close to that) and then it failed with the error message from above.

the exact error message would be a lot more helpful - snap a pic if easier

ciriousjoker, 2025-02-12:
[image]

MrChromebox, 2025-02-12:
looks to me like ChromeOS beta is breaking something. The commands on those lines are:

```
$CURL -sLO "${rwlegacy_source}${rwlegacy_file}.md5"
$CURL -sLO "${rwlegacy_source}${rwlegacy_file}"
```
`CURL` is defined in the firmware-util script (as either `curl` or `curl -k`). It's also used in various other places in the script, and has been for years now, without any issues. For some reason, the shell thinks `CURL` is undefined and thus is replacing `$CURL` with nothing. I don't think this is an error with my script.

ciriousjoker, 2025-02-12:
Tried it again, now it works. Now that I think of it, some other stuff on the system was also weird: battery kept displaying "charging" even when unplugged, battery percentage was fixed to 67% with 52min left. Trying to turn it off resulted in white screen freeze.

No clue what's going on here, but since it works now, this can be closed.

ciriousjoker, 2025-02-12:
Thanks a lot for your help!

closed 2025-02-12 by ciriousjoker