MrChromebox/firmware #726: Bricked hp x360 11 G3 EE

evloebenstein, 2025-02-22:
I first installed legacy Firmware directly to the eeprom with a ch341 and 1.8v module board without de soldering the chip but after seeing that the firmware didn't flash correctly i decided to de solder the eeprom and flash it directly to it .
This managed to flash the rom but when I soldered it back and turned on the Chromebook it showed a black screen. 
First I thought it was my soldering so I decided to check it and also the eeprom to see if it didn't burn or touched some resistance nearby.
The chip was still working, it read fine and I also managed to rollback the rom But it still won't show image.

evloebenstein, 2025-02-23:
I flashes the file coreboot_edk2-vortininja-mrchromebox_20240914.rom via a Windows Flasher Neo peogrammer

No command line used 
Though I also tried it again using this https://docs.mrchromebox.tech/docs/support/unbricking/unbrick-ch341a.html

evloebenstein, 2025-02-23:
I had problems flashing the rom with the program I was using but the solution I found was wiping the ic completely and then flashing.
It worked but I still shows a black screen

MrChromebox, 2025-02-23:
I don't have any reason to suspect there's an issue with the firmware. Why did you flash the device using an external programmer to begin with?

evloebenstein, 2025-02-23:
I saw a video using that software to flash the rom that way 
https://youtu.be/dceYuO3Mxuk?si=8BANNAig7-EolX5w
So I gave it a try. It didn't do anything at first, because after flashing I  turned on the Chromebook it still had the stock bios with a enterprise enrollment.

Then I saw your page about using a Linux machine to flash it and it happened again the same thing.

Now I am trying to use the unbricking tutorial again because I see that before flashing I need to clear the ic so it flashes correctly.

evloebenstein, 2025-02-23:
Flashes correctly with the vpd and hwid but it still does not show image

evloebenstein, 2025-02-23:
Maybe screen drivers? 

Is there some way to put the legacy drivers on the software?

evloebenstein, 2025-02-23:
[Stock ROM and coreboot modified.zip](https://github.com/user-attachments/files/18927720/Stock.ROM.and.coreboot.modified.zip)
Here is a copy of the stock rom and the one I did with the tunbricking tutorial if it helps

MrChromebox, 2025-02-23:
> Is there some way to put the legacy drivers on the software?

I don't know why you keep using this term legacy, nor do I know what you mean by it.

> Here is a copy of the stock rom and the one I did with the tunbricking tutorial if it helps

can you flash the stock ROM back? does the device boot with it?

closed 2026-11-04 by MrChromebox