MrChromebox/firmware #731: Lenovo Chromebook C340-11

Hexseas, 2025-03-07:
My screen froze so i reboot my chromebook
but when I rebooted it says: "Booting from 'Ubuntu' failed; verify it contains a 64 bit UEFI OS"
Exiting from there takes me to UEFI interactive shell
Now a reboot will have the rabbit flash on and stay there while the GNU GRUB command line menu overlays on top of it.
A default reboot goes straight to the GRUB command line menu
And I don't see any ubuntu on the Chrultrabook Docs where it only says for the bootable USB. "keep in mind Ubuntu and Ubuntu-based derivatives are not supported."
How am I to approach this?

MrChromebox, 2025-03-07:
boot to the UEFI settings, then go to boot menu. What are all of the options listed (in order)?

Hexseas, 2025-03-08:
Ubuntu
Ubuntu 
eMMC device
UEFI Shell
USB: Generic USB storage (I don't have a USB plugged in, only a 256 GB small chip [forgot it's name lol])

Hexseas, 2025-03-08:
No, same thing as booting from Ubuntu, "failed; verify it contains a 64 bit UEFI OS"

Hexseas, 2025-03-08:
Do I need to reinstall Ubuntu? Since it can't reboot anymore?

MrChromebox, 2025-03-08:
well, either the bootloader is corrupted, or your eMMC is on its way out. You can boot a live USB and repair/reinstall the bootloader without reinstalling Ubuntu

Hexseas, 2025-03-08:
How would I do that, not very well versed in this. Do I just plug in a USB? Is there a website or guide on how to do this?

MrChromebox, 2025-03-09:
boot from a ubuntu USB and chroot into your existing install and reinstall grub. plenty of guides abound

Hexseas, 2025-03-09:
Chroot? So reinstalling via the boot menu with an Ubuntu formatted USB? Is grub installation seperate thing,?

MrChromebox, 2025-03-09:
there are plenty of articles on how to do so, such as:
https://support.system76.com/articles/bootloader/

or you can just reinstall.

Hexseas, 2025-03-09:
Thanks for the help, you can consider this closed and I'll open a new one if it doesn't work

closed 2025-03-09 by Hexseas