MrChromebox/firmware #733: Chromebook (Acer C933 Droid) not recognizing arch linux bootable USB

3nndo, 2025-03-16:
I was triying to change from Debian to Arch, but the bios doesn't seem to detect the usb, it doesn't appear in Boot Menu and neither in Boot Manager.

[image]

[image]

3nndo, 2025-03-16:
I tired using Balena and Ventoy (Although im nopt so sure how to use last one) and it just doesn't appear

MrChromebox, 2025-03-16:
as per the FAQ, it has nothing to do with what is on the USB or how you wrote it. Try the mitigations listed in the FAQ or use another USB stick.

3nndo, 2025-03-16:
> as per the FAQ, it has nothing to do with what is on the USB or how you wrote it. Try the mitigations listed in the FAQ or use another USB stick.

Where can I find the FAQ?

MrChromebox, 2025-03-16:
> Where can I find the FAQ?

on my site, along with all the rest of the documentation

closed 2026-11-04 by MrChromebox