MrChromebox/firmware #734: Low amount of VRAM on Careena chromebook

eepylilyy, 2025-03-16:
Running Arch Linux on a hp chromebook 14 db0031nr (careena). It only shows 32 mb vram on LACT. I know its like that by default but I feel like 128 or 256mb would be better. I would really appreciate it if you would increase it. Videos seem to be sluggish a lot. So I think it would benefit overall performance if VRAM was increased to 128mb.

MrChromebox, 2025-03-16:
this is something you can recompile and change with a simple config setting, but I will bump it for the next release

eepylilyy, 2025-03-17:
Could I ask which config setting? I don't really know much about coreboot or edk2. If it's too much of a hassle to explain then don't bother.

MrChromebox, 2025-03-17:
when running menuconfig, it's under mainboard:

[image]

or if adding to the config file, you'd add `CONFIG_KAHLEE_UMA_SIZE_128MB=y`

eepylilyy, 2025-03-17:
Hate to keep bothering you like this but do I just compile coreboot from the official source code like using git clone https://review.coreboot.org/coreboot.git or do you have your own version of it? Again, I already really appreciate the help you've given me so far but I just need this cleared up. Thanks again!

MrChromebox, 2025-03-17:
you would clone my repo, build the toolchain, edit the config file per above, then build using my build script.

or just flash the build attached here

[coreboot_edk2-careena-mrchromebox_20250317.zip](https://github.com/user-attachments/files/19288566/coreboot_edk2-careena-mrchromebox_20250317.zip)

eepylilyy, 2025-03-17:
Okay, I tried your build and it works perfectly. I tried doing it on my own and did end up making it work but it was a bit janky. Oneee last favor though could you make me a 256mb build? If that's too much trouble then you don't have to

MrChromebox, 2025-03-17:
I'm not sure you're going to see much of a difference between 128MB and 256MB due to the way UMA is used on these devices