MrChromebox/firmware #736: TEEMO: fan speed spikes on wakeup and reboot

bam80, 2025-03-20:
It seems fan spins outs to the maximum for a short period of time after wakeup from suspend and on reboot.

**     Fw Ver: 24.12-969-ge94e7469784 (03/17/2025)

MrChromebox, 2025-03-20:
I'm not seeing this here on either of my i7 TEEMO boxes when using the EC-RO firmware. The fan is on and audible, but nowhere near maximum

bam80, 2025-03-20:
I'm on EC-RO too. Please note the spikes duration is only 2-3 seconds after wakeup, but they are clearly audible.

MrChromebox, 2025-03-20:
I'm not seeing that here - not much can be done anyway, as the spike is before coreboot switches the fan mode to auto, and we're dealing with the RO EC

MrChromebox, 2025-03-20:
updating EC-RO is very risky, and I don't have any tooling to do it or recover from it.

MrChromebox, 2026-10-27:
The only time the fan surges is when the device starts from the RO firmware (ie, power removed). If the device is simply powered down and the EC stays in RW, then there's no issue. Not going to risk changing the RO firmware for a half second fan surge.

bam80, 2026-10-27:
Just tested it again, you are right:
it was power outage here, and after that the first power on was with the fan spike.
On subsequent power off/on, there was no issue.
Thanks.

MrChromebox, 2026-11-05:
Closing as expected behavior: brief fan surge only when starting from EC-RO after power loss; normal power off/on with EC in RW is fine. Not worth the risk of an EC-RO update for this.

closed 2026-11-05 by MrChromebox