MrChromebox/firmware #739: Battery drain even after laptop is off

Tu2l, 2025-03-29:
Hi @MrChromebox,

Since the last few weeks, I've been noticing strange behavior with my Morphious Chromebook. I remember turning off my laptop with a battery percentage above 30%, but after turning it on after a few days, the battery percentage is always below 20%. I'm not sure this issue is related to firmware or Fedora 41. 

Chromebook specs - 
Lenovo C13 Yoga
CPU - R7 3700C (originally it was R5 but due to some issue the MOBO was replaced, with that CPU was also upgraded)
Memory - 16GiB

Thanks.

MrChromebox, 2025-03-29:
I'll see if I can reproduce here on my morphius

andy-shev, 2025-06-05:
I have Pixel (Samus) which suffers from the same bug. Power Off (either in OS or as an HW button even before bootloader starts behaves like a cold reset. In OS the only option is to Suspend (to RAM) which is not okay at all. This is a serious bug here.

andy-shev, 2025-06-05:
> I'm not able to reproduce this on either Morphius or Samus. Some battery drain is expected over a few days even when the device is fully powered off.

FWIW, which version (of MrChromeBox) you tested? The 2503?

MrChromebox, 2025-06-05:
> FWIW, which version (of MrChromeBox) you tested? The 2503?

yes, latest public release

MrChromebox, 2025-06-08:
> Ah, I didn't know that (it's actually my first time ever I put my hands on Chromebook HW as an (advanced) user). But I believe it may be changed via hacking the EC firmware.

it should be able to be changed in coreboot as well, since it sets what are valid wake events.

> The OS (Linux) prints in dmesg that the INT3430 device is in PIO mode. But again, I know how to fix that.

the GSPI0 device for SAMUS should probably be off anyway, it's not needed. I'll change that for the next release.

andy-shev, 2025-06-08:
> > Ah, I didn't know that (it's actually my first time ever I put my hands on Chromebook HW as an (advanced) user). But I believe it may be changed via hacking the EC firmware.
> 
> it should be able to be changed in coreboot as well, since it sets what are valid wake events.

If it's an official feature, no need to disable it. It's rather a surprise to me, but whatever...

> > The OS (Linux) prints in dmesg that the INT3430 device is in PIO mode. But again, I know how to fix that.
> 
> the GSPI0 device for SAMUS should probably be off anyway, it's not needed. I'll change that for the next release.

Hmm... Since it's enabled, it's most likely might be used. Have you had a schematics of it to check?

Tu2l, 2025-07-04:
@MrChromebox still facing the same issue; it feels like the system is never turning off completely. Today I tried to turn on the system (after 3-4 days) by opening the lid. To my surprise, the battery was completely drained. I had to plug it into the charger.

Tu2l, 2025-07-09:
@MrChromebox Today I faced the same situation but in worse condition. Yesterday I turned off my system with approximately 40% charge left. Today I tried to turn it on and realized the system was completely dead. I plugged in the charger, the LEDs blinked, the screen turned on, and then it turned off again. This cycle keeps repeating as long as the charger is plugged in. One thing I realized is the system is not capable of taking charge while it is off or the battery is completely drained; it tries to boot up as soon as the charger is plugged in. I'm able to revive the system by taking off the battery, then turning on the system with the charger connected, and then connecting the battery while the system is on. Now it's charging, but I fear the same thing might repeat in the future. I'm not 100% sure if the battery drain issue is related to the firmware or not, but I strongly believe that not being able to charge while the system is off or booting up as soon as the charger is connected is related to the firmware, so please look into this if possible.

MrChromebox, 2025-07-10:
@Tu2l this sounds like there is some issue with the battery itself. I've never seen such behavior here and don't see how this could be coreboot related

t2nel, 2025-07-29:
Unlike hibernated or shut down windows laptops, you can't just turn off chromebook and expect it have a similar battery a week later. It is still using a bit of power even when you turn it off or hibernate.

To _completely_ shut down and keep battery level long term, you have to use this method:
https://support.google.com/chrome/a/answer/9139543?hl=en

This will disable lid wakeup and also power button. Connecting a charger will turn it on again.