MrChromebox/firmware #740: No UEFI Boot entries shown after Linux installation (RW_LEGACY)

akuehntopf, 2025-03-29:
Device: Asus Chromebook Plus (MARASOV)
Firmware Version: rwl_altfw_adl-mrchromebox_20250313.bin

Steps taken:
---------------------
1. Enable Developer Mode
2. Install Firmware on VT-2 using "Install/Update RW_LEGACY firmware"
3. Resize stock partitions using the "CRAP" tool
4. Reboot
5. Install Linux Distribution

Problem description:
--------------------------------
Installation is working fine but afterwards the edk2 UEFI Menu is always empty, the only option i can select is "UEFI Shell".

I tried with Fedora 41, Ubuntu 24.10 and Arch Linux.

For partition layout I used a custom layout as stated by the docs. 
I used Partition #12 as EFI and added another partitions as demanded by the installed OS.

This used to work before as I already had a running Fedora 41 on this machine a few weeks before. I had reverted to stock in the meantime, now I can't get any Linux Distro to boot.

Any help would be appreciated!

MrChromebox, 2025-03-29:
the only UEFI entries which exist in the boot menu for RWL are fixed disks. What type of internal storage does your device use?

akuehntopf, 2025-03-29:
It's UFS

akuehntopf, 2025-03-29:
It's UFS.
The device is an ASUS CX34, to be exact.

MrChromebox, 2025-03-29:
let me rebuild the RWL payload and make sure UFS is enabled

MrChromebox, 2025-03-29:
the problem with UFS is that if I enable it universally for ADL devices, the RWL payload locks up if no UFS present -- or at least that's what I'm seeing under emulation. I'll have to figure out a good way to detect which version is needed

MrChromebox, 2025-03-29:
ok, I pushed a RWL update for ADL that has UFS support, give it a try and let mw know if it works for you. I tested it on Banshee w/NVMe and it booted fine

akuehntopf, 2025-03-30:
Thanks for looking into this @MrChromebox.
Happy to confirm that it's working, boot entries appeared immediately after installing the updated RW_LEGACY fw.

Boot entries show up as:

- UEFI Shell
- Micron
- Micron 2