MrChromebox/firmware #747: Hardware operation failures caused by issues related to BIOS device initialization and ACPI.

BahamutLee, 2025-04-13:
Hello, I sincerely appreciate the remarkable efforts made by your team. These efforts have benefited countless Chromebook enthusiasts like myself. Thank you very much.

My testing device: HP Pro c640 G2 Chromebook.

Hardware configuration:
CPU: Intel 11th Generation Core i5-1145G7
Motherboard: Google Elemi
Motherboard version number: Model: 0GCC DA0GCCMBAD0 REV: D
EMMC: Genesys Logic, Device ID: E763
TF card reader: Genesys Logic, Device ID: 9750

        Problem: There may be issues such as power device and hardware initialization errors during the BIOS startup process, and it is suspected that there are errors in the ACPI aspect.

Specific situations:
        1. Restart after the screen goes black when shutting down: After the computer has been in the shutdown state for several hours, when shutting down for the first time, the machine will automatically restart once after the screen turns off. After that, it can be shut down normally when shutting down again. When shutting down for the first time after inserting a TF card, the shutdown will definitely turn into a restart. Multiple devices of the same type have been checked, and this problem occurs in all of them. The driver issues installed after entering Windows have also been investigated, but the problem cannot be solved.
        2. Conflict between the TF card and the Emmc device: After inserting the TF card, the corresponding TF card hardware in the Device Manager will most likely display a yellow exclamation mark, and at the same time, an IO error will be shown in the Windows system log. Occasionally, after inserting the TF card, the Emmc device will show an exclamation mark and report an IO error.
        3. The Type-c interface cannot be used: Only by inserting a Type-c device before booting up can the Type-c device run normally. Otherwise, if the Type-c device is not inserted before starting up, there will be no response when the Type-c device is connected.
        4. When installing the Windows system using the EMMC, an error will be reported when the installation process reaches about 40%, indicating that an error has occurred and the installation cannot proceed. This problem does not occur when installing on an NVME hard drive.

ellyq, 2025-05-22:
1. I have no idea what you're trying to say in this paragraph.
2. Please check in Linux. ELEMI uses the same baseboard as ELDRID, I did not experience any issues with mSD (although ELDRID uses NVME). Post logs: https://docs.chrultrabook.com/docs/debugging/
3. Use up-to-date distribution, read: https://docs.chrultrabook.com/docs/installing/post-install.html#fixing-usb-c-on-tigerlake-and-alderlake

BahamutLee, 2025-05-27:
> 1. I have no idea what you're trying to say in this paragraph.
> 2. Please check in Linux. ELEMI uses the same baseboard as ELDRID, I did not experience any issues with mSD (although ELDRID uses [...]

> I would like to report some bugs observed on my machine.
>1、 The original motherboard only has a TF card slot and eMMC, with no Nvme interface. I added an Nvme socket along with the corresponding power supply and logic controller.
> 2、After flashing coreboot on the original motherboard, installing Windows on the eMMC encounters errors midway, while installation on Nvme works normally.
> 3、There is a probability that inserting a TF card causes the system to automatically reboot after a Windows shutdown.
> Overall, these issues are quite strange, especially the problems with unexpected reboots and unexplained shutdowns. I hope to find others facing similar situations to discuss this together. I have a total of 5 units of this laptop model, and each of them exhibits the same issues.

rosanyadav08, 2025-08-05:
On lenovo chromebook c13 yoga its directly showing something went wrong and not is working 

I tried everything but its shows the same any option please guide[image]

Can anyone help me with this