MrChromebox/firmware #748: chromebook works for 1 second then it shuts off  immediately

AR8XPRO, 2025-04-18:
chromebook works for 1 second then it shuts off  immediately, i tried charging it WHAT DO I DO
(btw i installed the frimware 3 weeks ago and then this happend today)

MrChromebox, 2025-04-18:
I have no idea, what makes you think this is a firmware issue?

AR8XPRO, 2025-04-19:
idk but the device dosent work now simply(i used fedora first, it worked ,reinstalled fedora again*idk why* but the device still works
i used lubuntu it worked fine with severe issues ,but it still worked then i installed ubuntu 22.04 i had sound issues then installed ubuntu 24.04 the sound fixed and BOOM after only 5 days this happened and i idk what to do now)

MrChromebox, 2025-04-19:
not sure what to tell you, this isn't a firmware issue if the device was working and then just stopped

AR8XPRO, 2025-04-20:
so.... i dont know what to do yet

MrChromebox, 2025-04-20:
there are plenty of other resources on the internet for dealing with Chromebook-related issues. This github tracker is for issues with my firmware, not general troubleshooting or hardware issues.

closed 2025-04-20 by MrChromebox