MrChromebox/firmware #750: HP Chromebook x360 14b (Blooguard) many problems after 2503 update

moakt3, 2025-04-29:
Hi.

HP Chromebook x360 14b (Blooguard)

Today I`m install 2503 firmware update. After installation and reboot in Windows 24H2 stop working touchpad and partially keyboard. Windows reinstall not help. Firmware reinstall not help to. I`m install Devuan and everything seems to be fine. But to many startup errors in boot log:
[code]ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PCI0.GPCB], AE_NOT_FOUND (20240827/psargs-332)
ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PCI0.CNVW._SB.PCI0.PS2K], AE_NOT_FOUND (20240827/dswload2-162)
ACPI Error: AE_NOT_FOUND, During name lookup/catalog (20240827/psobject-220)
ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PCI0.CNVW._SB.PCI0.LPCB.EC0.CREC.CKSC], AE_NOT_FOUND (20240827/dswload2-162)
ACPI Error: AE_NOT_FOUND, During name lookup/catalog (20240827/psobject-220)[/code]

Can you give me a link to [b]coreboot_edk2-blooguard-mrchromebox_20240914.rom[/b]? I try manually reflash old firmware.

MrChromebox, 2025-04-29:
please provide the full dmesg log from Devuan

moakt3, 2025-04-29:
I just downgrade to 2409.1. If you need i install 2503 again.

PS: When i use mrchromebox rom to downgrade do i need hwid file?

MrChromebox, 2025-04-29:
yes, I need the log to see what's wrong.

HWID is only needed when going back to stock and you're a dummy and lose your stock backup.

moakt3, 2025-04-29:
Sorry it took so long. Removed sensitive data. In the attached file.

> [    0.000000] Linux version 6.12.12+bpo-amd64 (debian-kernel@lists.debian.org) (x86_64-linux-gnu-gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT_DYNAMIC Debian 6.12.12-1~bpo12+1 (2025-02-23)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.12.12+bpo-amd64 root=XXXXXXXXXXXXXXXXXXXXXXXXX ro ipv6.disable=1 quiet
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000000fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000001000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000000cfa5fff] usable
[    0.000000] BIOS-e820: [mem 0x000000000cfa6000-0x000000000d0b9fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000d0ba000-0x000000000fdd9fff] usable
[    0.000000] BIOS-e820: [mem 0x000000000fdda000-0x000000000ffd9fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000ffda000-0x000000000ffecfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000ffed000-0x000000000fffefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000000ffff000-0x000000000fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000010000000-0x0000000012150fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000012151000-0x00000000799e3fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000799e4000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffad1000-0x00000000ffb10fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000017fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] APIC: Static calls initialized
[    0.000000] efi: EFI v2.7 by EDK II
[    0.000000] efi: SMBIOS=0xffc1000 SMBIOS 3.0=0xffbf000 ACPI=0xfffe000 ACPI 2.0=0xfffe014 MEMATTR=0xccde018 MOKvar=0xff73000 INITRD=0xc664c98 
[    0.000000] efi: Remove mem91: MMIO range=[0xffad1000-0xffb10fff] (0MB) from e820 map
[    0.000000] e820: remove [mem 0xffad1000-0xffb10fff] reserved
[    0.000000] Kernel is locked down from EFI Secure Boot; see man kernel_lockdown.7
[    0.000000] secureboot: Secure boot enabled
[    0.000000] SMBIOS 3.0.0 present.
[    0.000000] DMI: Google Blooguard/Blooguard, BIOS MrChromebox-2503.0 04/27/2025
[    0.000000] DMI: Memory slots populated: 4/4
[    0.000000] tsc: Detected 1094.400 MHz processor
[    0.000022] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000025] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000033] last_pfn = 0x180000 max_arch_pfn = 0x400000000
[    0.000039] MTRR map: 4 entries (3 fixed + 1 variable; max 23), built from 10 variable MTRRs
[    0.000042] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000110] last_pfn = 0x799e4 max_arch_pfn = 0x400000000
[    0.004217] Using GB pages for direct mapping
[    0.005113] RAMDISK: [mem 0x07db7000-0x0abf6fff]
[    0.005116] ACPI: Early table checksum verification disabled
[    0.005121] ACPI: RSDP 0x000000000FFFE014 000024 (v02 COREv4)
[    0.005127] ACPI: XSDT 0x000000000FFFD0E8 00006C (v01 COREv4 COREBOOT 00000000      01000013)
[    0.005134] ACPI: FACP 0x000000000FFFC000 000114 (v06 COREv4 COREBOOT 00000000 CORE 20230628)
[    0.005141] ACPI: DSDT 0x000000000FFF8000 0036B8 (v02 COREv4 COREBOOT 20110725 INTL 20230628)
[    0.005146] ACPI: FACS 0x00000000799FC240 000040
[    0.005149] ACPI: SSDT 0x000000000FFF6000 001967 (v02 COREv4 COREBOOT 00000000 CORE 20230628)
[    0.005153] ACPI: MCFG 0x000000000FFF5000 00003C (v01 COREv4 COREBOOT 00000000 CORE 20230628)
[    0.005157] ACPI: TPM2 0x000000000FFF4000 00004C (v04 COREv4 COREBOOT 00000000 CORE 20230628)
[    0.005161] ACPI: APIC 0x000000000FFF3000 000072 (v03 COREv4 COREBOOT 00000000 CORE 20230628)
[    0.005165] ACPI: NHLT 0x000000000FFF2000 000E64 (v05 COREv4 COREBOOT 00000000 CORE 00000000)
[    0.005169] ACPI: DBG2 0x000000000FFF1000 000061 (v00 COREv4 COREBOOT 00000000 CORE 20230628)
[    0.005172] ACPI: HPET 0x000000000FFF0000 000038 (v01 COREv4 COREBOOT 00000000 CORE 20230628)
[    0.005176] ACPI: BGRT 0x000000000FFEF000 000038 (v01 INTEL  EDK2     00000002      01000013)
[    0.005179] ACPI: Reserving FACP table memory at [mem 0xfffc000-0xfffc113]
[    0.005181] ACPI: Reserving DSDT table memory at [mem 0xfff8000-0xfffb6b7]
[    0.005182] ACPI: Reserving FACS table memory at [mem 0x799fc240-0x799fc27f]
[    0.005184] ACPI: Reserving SSDT table memory at [mem 0xfff6000-0xfff7966]
[    0.005185] ACPI: Reserving MCFG table memory at [mem 0xfff5000-0xfff503b]
[    0.005186] ACPI: Reserving TPM2 table memory at [mem 0xfff4000-0xfff404b]
[    0.005187] ACPI: Reserving APIC table memory at [mem 0xfff3000-0xfff3071]
[    0.005188] ACPI: Reserving NHLT table memory at [mem 0xfff2000-0xfff2e63]
[    0.005190] ACPI: Reserving DBG2 table memory at [mem 0xfff1000-0xfff1060]
[    0.005191] ACPI: Reserving HPET table memory at [mem 0xfff0000-0xfff0037]
[    0.005192] ACPI: Reserving BGRT table memory at [mem 0xffef000-0xffef037]
[    0.005355] No NUMA configuration found
[    0.005357] Faking a node at [mem 0x0000000000000000-0x000000017fffffff]
[    0.005373] NODE_DATA(0) allocated [mem 0x17ffd5680-0x17fffffff]
[    0.005758] Zone ranges:
[    0.005759]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[... 637 lines cut ...]
[    5.559577] Bluetooth: hci0: Waiting for firmware download to complete
[... cut ...]
[   10.112437] da7219 i2c-DLGS7219:00: SRM failed to lock
[   10.115397] wlan0: RX AssocResp from XXXXXXXXXXXXXXXXXXX (capab=0x411 status=0 aid=1)
[   10.121245] wlan0: associated
[   10.287010] Lockdown: elogind-daemon: hibernation is restricted; see man kernel_lockdown.7
[   10.357953] Lockdown: elogind-daemon: hibernation is restricted; see man kernel_lockdown.7
[   10.364495] Lockdown: elogind-daemon: hibernation is restricted; see man kernel_lockdown.7
[   11.216474] da7219 i2c-DLGS7219:00: SRM failed to lock

MrChromebox, 2025-04-30:
ok, fixed builds incoming

MrChromebox, 2025-04-30:
fixed

closed 2025-04-30 by MrChromebox

moakt3, 2025-04-30:
Thank YOU. Very prompt. Already installed.