MrChromebox/firmware #751: ASUS CX3402CB ("MARASOV") - Something went wront launching the alternate bootloader

AlphaUrsaeMinoris, 2025-04-30:
Steps taken:
- Enable developer mode
- Install/Update RW_LEGACY firmware
- Reboot
- Select edk2

Problem:
"There was an error while starting the alternative bootloader" (translated from German, the exact wording might be different on an English system)
In the log: "segment type 45444f43 not implemented. Exiting"

What I tried so far:
Using the older 20250313 firmware - makes no difference.

What additional information can I provide to help you find the issue?

MrChromebox, 2025-04-30:
none, I just fixed it -- re-flash and LMK if it's good now.

All Alderlake boards had broken RWL originally, and one at a time Google seems to be fixing the original bug, which I had worked around. So now I need to undo that workaround (a special RWL build) as boards get fixed.

AlphaUrsaeMinoris, 2025-04-30:
That was it! Thank you a lot! I really appreciate your work.

closed 2025-04-30 by AlphaUrsaeMinoris