MrChromebox/firmware #756: General Protection Fault on Craasneto (Acer Chromebook 314 (CB314-4H-C90V))

Cydox, 2025-05-05:
I couldn't get Craasneto to boot with the full UEFI firmware. It would turn on the screen backlight after ~1.5 minutes but stay black.

So far I tried three different firmwares. Behavior is always the same:
1. self compiled from 2408 (to add most recent microcode)
2. prebuilt 2503 (https://www.mrchromebox.tech/files/firmware/full_rom/coreboot_edk2-craasneto-mrchromebox_20250427.rom)
3. self compiled 2503 (to add serial output)

I compiled 2503 from source with `CONFIG_CONSOLE_SERIAL=y` added to get serial output (full log attached as file). Here's the relevant part with the general protection fault. It appears to be happening right after jumping to edk2.
https://github.com/MrChromebox/coreboot/blob/238cd45199732081d1ce8ee0199b6604373a2121/src/lib/prog_loaders.c#L195

```
[DEBUG]  BS: BS_WRITE_TABLES exit times (exec / console): 169 / 331 ms
[INFO ]  CBFS: Found 'fallback/payload' @0x179680 size 0x15e5b7 in mcache @0x76c0d30c
[DEBUG]  Checking segment from ROM address 0xffdcd8ac
[DEBUG]  Checking segment from ROM address 0xffdcd8c8
[DEBUG]  Loading segment from ROM address 0xffdcd8ac
[DEBUG]    code (compression=1)
[DEBUG]    New segment dstaddr 0x00800000 memsize 0x800000 srcaddr 0xffdcd8e4 filesize 0x15e57f
[DEBUG]  Loading Segment: addr: 0x00800000 memsz: 0x0000000000800000 filesz: 0x000000000015e57f
[DEBUG]  using LZMA
[DEBUG]  Loading segment from ROM address 0xffdcd8c8
[DEBUG]    Entry Point 0x00801714
[DEBUG]  BS: BS_PAYLOAD_LOAD run times (exec / console): 996 / 65 ms
[INFO ]  coreboot skipped calling FSP notify phase: 00000040.
[INFO ]  coreboot skipped calling FSP notify phase: 000000f0.
[DEBUG]  BS: BS_PAYLOAD_LOAD exit times (exec / console): 0 / 14 ms
[DEBUG]  Finalizing chipset.
[DEBUG]  apm_control: Finalizing SMM.
[DEBUG]  APMC done.
[INFO ]  CSE: EOP requested action: continue boot
[WARN ]  HECI: CSE device 16.1 is disabled
[WARN ]  HECI: CSE device 16.2 is disabled
[WARN ]  HECI: CSE device 16.3 is disabled
[WARN ]  HECI: CSE device 16.4 is disabled
[WARN ]  HECI: CSE device 16.5 is disabled
[DEBUG]  BS: BS_PAYLOAD_BOOT entry times (exec / console): 0 / 43 ms
[DEBUG]  mp_park_aps done after 0 msecs.
[DEBUG]  Jumping to boot code at 0x00801714(0x769ff000)
!!!! X64 Exception Type - 0D(#GP - General Protection)  CPU Apic ID - 00000008 !!!!
ExceptionData - 0000000000000000
RIP  - 00000000767B82CE, CS  - 0000000000000038, RFLAGS - 0000000000010006
RAX  - 00000000000000AF, RCX - AFAFAFAFAFAFAFAF, RDX - 000000007696F801
RBX  - AFAFAFAFAFAFAFCC, RSP - 0000000004FDC498, RBP - 0000000004FDC540
RSI  - 0000000000000000, RDI - AFAFAFAFAFAFAFAF
R8   - 8000000000000000, R9  - 0000000004FDC5D0, R10 - 0000000000000000
R11  - 0000000004FF7778, R12 - 0000000076309418, R13 - 00000000000000AF
R14  - 000000000000003C, R15 - 0000000000000000
DS   - 0000000000000030, ES  - 0000000000000030, FS  - 0000000000000030
GS   - 0000000000000030, SS  - 0000000000000030
CR0  - 0000000080010013, CR2 - 0000000000000000, CR3 - 0000000004C01000
CR4  - 0000000000000668, CR8 - 0000000000000000
DR0  - 0000000000000000, DR1 - 0000000000000000, DR2 - 0000000076AE4B20
DR3  - 0000000000000000, DR6 - 00000000FFFF0FF0, DR7 - 000000000D0F0400
GDTR - 0000000076996000 0000000000000047, LDTR - 0000000000000000
IDTR - 0000000076331018 0000000000000FFF,   TR - 0000000000000000
FXSAVE_STATE - 0000000004FDC0F0
!!!! Find image based on IP(0x767B82CE) /home/jan/coreboot/payloads/external/edk2/workspace/Build/UefiPayloadPkgX64/RELEASE_COREBOOT/X64/MdeModulePkg/Universal/Variable/RuntimeDxe/VariableRuntimeDxe/DEBUG/VariableRuntimeDxe.dll (ImageBase=00000000767B6000, EntryPoint=0000000076830424) !!!!
```

Full log:
[serial_ap_log.txt](https://github.com/user-attachments/files/20041812/serial_ap_log.txt)

Cydox, 2025-05-05:
Just now saw that I need `CONFIG_EDK2_SERIAL_SUPPORT=y` to get serial output from edk2. Doing another flash with that added.

Cydox, 2025-05-05:
Adding `CONFIG_EDK2_SERIAL_SUPPORT=y` didn't change anything. No logging from edk2 prior to the crash.

MrChromebox, 2025-05-05:
> Adding CONFIG_EDK2_SERIAL_SUPPORT=y didn't change anything. No logging from edk2 prior to the crash.

you need:
`CONFIG_CONSOLE_SERIAL=y`
and
`CONFIG_EDK2_DEBUG=y`

serial output with an edk2 release build currently doesn't work

Cydox, 2025-05-05:
That worked. Here's the log:
[log.txt](https://github.com/user-attachments/files/20049515/log.txt)

Cydox, 2025-05-06:
Too add some more baseline info. The stock firmware runs without any issues.
Here's the git revision from the `revision` file: `3227377df6f`

I assume that revision is referring to what is in the RO section. Is there an easy way to get at the revision from the two firmwares in the RW section? When I'm booted using the stock firmware, the reported version is `Google_Craask.15217.718.0` (from `crossystem`)

MrChromebox, 2025-05-06:
>  Is there an easy way to get at the revision from the two firmwares in the RW section? 

`cbfstool coreboot.rom extract -r FW_MAIN_{A,B} -n revision -f revision.txt`

but those revisions point to google's repo, using google's blobs, etc. it's not particularly useful info

MrChromebox, 2025-05-06:
> That worked. Here's the log: [log.txt](https://github.com/user-attachments/files/20049515/log.txt)

sure looks to me like edk2 failed to init the NVRAM storage region set up by coreboot (SMMSTORE), which is not normal.

there's also a lot of image alignment crap, which means you should do a `make distclean` and rebuild to flush out some old edk2 settings.

I would add `# CONFIG_SMMSTORE is not set` to your config if you want to just get the device booting / verify that the NVRAM init is the issue.

Cydox, 2025-05-06:
> there's also a lot of image alignment crap, which means you should do a `make distclean` and rebuild to flush out some old edk2 settings.

Is the image alignment stuff actually fatal? I assume it will not be able to set protection flags in the appropriate way with stuff not aligned to page boundaries.

> 
> I would add `# CONFIG_SMMSTORE is not set` to your config if you want to just get the device booting / verify that the NVRAM init is the issue.

I'll try this first to see if that's the root cause.

Cydox, 2025-05-06:
> I would add `# CONFIG_SMMSTORE is not set` to your config if you want to just get the device booting / verify that the NVRAM init is the issue.

Yes NVRAM init is the issue. getting rid of CONFIG_SMMSTORE made it boot.

MrChromebox, 2025-05-06:
> Is the image alignment stuff actually fatal? I assume it will not be able to set protection flags in the appropriate way with stuff not aligned to page boundaries.

no, it just makes the log harder to read

> Yes NVRAM init is the issue. getting rid of CONFIG_SMMSTORE made it boot.

so now we are left with why is it failing on one device and working on another, when the two devices are using the same firmware an essentially identical

Cydox, 2025-05-06:
One more difference I discovered when reading the docs is that I did not run
`flashrom --wp-range 0 0`

Am I correct in thinking this only effects what range is protected if software write protect is enabled? So this shouldn't matter after `flashrom --wp-disable`

MrChromebox, 2025-05-06:
you don't need to run either of those commands, the script does it when flashing the UEFI Full ROM firmware, and it doesn't need to be done again unless you manually re-enable either option

Cydox, 2025-05-06:
I did not use the script to flash. I used `flashrom` directly. Sorry if this wasn't clear before.

`flashrom -p internal --ifd -i bios -w <rom_to_flash> -N`

Cydox, 2025-05-06:
Thank you so much for your help

closed 2025-05-06 by Cydox

MrChromebox, 2025-05-06:
and this is why I recommend people use the script (at least initially) :)

Cydox, 2025-05-06:
Fair point!