MrChromebox/firmware #757: QEMU: SD card is not visible by UEFI

bam80, 2025-05-07:
Hi Matt,
I'm preparing SD card installation image for my Asus Chromebox 3 with QEMU.

Surprisingly, OVMF UEFI doesn't see the card emulated, only SeaBIOS does.
I know the box itself doesn't have this problem, so maybe you could give me an idea what's going on here?
Thanks.

## Host environment
 - Operating system: Fedora 42           
 - OS/kernel version: 6.14.3-300.fc42.x86_64           
 - Architecture: x86_64               
 - QEMU flavor: qemu-system-x86_64                
 - QEMU version: qemu-9.1.2-3.fc41                
 - QEMU command line:
   ```
   qemu-kvm -bios /usr/share/OVMF/OVMF_CODE.fd -device sdhci-pci -drive if=none,file=Fedora-IoT-ostree-41-20241027.0.x86_64.iso,format=raw,id=MMC1 -device sd-card,drive=MMC1
   ```

## Emulated/Virtualized environment
 - Architecture: x86_64               

## Description of problem

SD card is not visible by OVMF UEFI, so it's not possible to boot from it:
```
UEFI Interactive Shell v2.2
EDK II
UEFI v2.70 (EDK II, 0x00010000)
Mapping table
     BLK0: Alias(s):
          PciRoot(0x0)/Pci(0x1,0x1)/Ata(0x0)
Press ESC in 1 seconds to skip startup.nsh or any other key to continue.
Shell>
```
It is visible by SeaBIOS though, if we remove the OVMF part from the commandline:
```
qemu-kvm -device sdhci-pci -drive if=none,file=Fedora-IoT-ostree-41-20241027.0.x86_64.iso,format=raw,id=MMC1 -device sd-card,drive=MMC1
```

MrChromebox, 2025-05-07:
I have no idea offhand, never having used OVFM. Have you tried using coreboot+edk2payload with qemu? I know that boots at least. A debug build with a full serial log would be a good start.

bam80, 2025-05-07:
Thanks, IDK about edk2payload, but apparently someone already faced an issue and made a log:
https://edk2.groups.io/g/bugs/message/6548

MrChromebox, 2026-11-05:
Closing: this is stock QEMU/OVMF SDHCI emulation behavior, not MrChromebox Chromebook firmware. The Teemo/box itself isn’t involved. For QEMU testing, coreboot+edk2 payload (or SeaBIOS) is a better stand-in than distro OVMF.

closed 2026-11-05 by MrChromebox