MrChromebox/firmware #758: Sometimes USB devices reconnected (fallof) or not initialized (in coreboot menu also).

moakt3, 2025-05-17:
I`m sorry if this is not a firmware bug problem.

I have WD external HDD, Transcend USB flash 256Gb and SanDisk USB flash 128Gb. All this devices use USB 3.2 Type-A. Sometimes in Windows, Linux and even coreboot menu they may reconnect (fallof) or not initialized. Sometimes happens low write speed (USB flash only). Why this happens? Cheap chipset inside or firmware bug?

PS: Two USB type-C works fine.

MrChromebox, 2025-05-18:
if the issue were only in the firmware boot menu, I'd say it's an issue with the edk2 part of firmware. But since you said the issue exists in Linux/Windows as well, that would suggest something either with the port tuning in coreboot or a potential hardware issue

moakt3, 2025-05-19:
I understand. Thank you for your time.

closed 2025-05-19 by moakt3