MrChromebox/firmware #761: CEC hardware not working. Guidance request.

nariox, 2025-06-01:
Hi MrChromebox,

I have recently purchased a CTL Chromebox CBx1. It is one of the Kaby Lake Chromeboxes that supports HDMI-CEC. Although the hardware should support it, the driver (`cros-ec-cec`) is not correctly detecting it, giving me this message on `dmesg`:
```[    7.000795] cros-ec-cec cros-ec-cec.2.auto: CEC notifier not configured for this hardware```
After a bit of digging, I realized that this comes from the driver not finding a match on this table: [LINK](https://github.com/torvalds/linux/blob/cd2e103d57e5615f9bb027d772f93b9efd567224/drivers/media/cec/platform/cros-ec/cros-ec-cec.c#L304-L337) So, apparently, it is looking for "Fizz" on the `product_name`, instead of `Wukong`. Now, I don't know if this is the kind of thing that should be changed in the driver or in the firmware. I have added a line on the table and recompiled the module. With that I get the driver to work and create `/dev/cec0`:
```[ 3990.648562] cros_ec_cec: loading out-of-tree module taints kernel.
[ 3990.648570] cros_ec_cec: module verification failed: signature and/or required key missing - tainting kernel
[ 3990.652290] Registered IR keymap rc-cec
[ 3990.652341] rc rc0: cros-ec-cec as /devices/pci0000:00/0000:00:1f.0/PNP0C09:00/GOOG0004:00/cros-ec-dev.1.auto/cros-ec-cec.2.auto/rc/rc0
[ 3990.652406] input: cros-ec-cec as /devices/pci0000:00/0000:00:1f.0/PNP0C09:00/GOOG0004:00/cros-ec-dev.1.auto/cros-ec-cec.2.auto/rc/rc0/input13
```
So, that said, what would be your expert opinion? Should I write a patch for the kernel driver? Or is this something that belongs as change in the firmware?

P.S.: thank you for maintaining this project. I am very thankful.

MrChromebox, 2025-06-02:
This is an unfortunate side effect of the way Linux does device matching. My preference would be to have Linux match all of the Kabylake Chromeboxes by name, or look at the Family field ('Google_Fizz') in addition to the vendor and product name.

nariox, 2025-06-26:
Patch sent, included just the Fizz boards for now. If devs are amenable to the changes, we can add the remainder of the boards. Otherwise I could suggest a different matching scheme (like Family instead of Product Name).

Link here: https://www.spinics.net/lists/linux-media/msg279805.html

movr4x, 2025-06-30:
@nariox 

> Patch sent, included just the Fizz boards for now. If devs are amenable to the changes, we can add the remainder of the boards. Otherwise I could suggest a different matching scheme (like Family instead of Product Name).
> 
> Link here: https://www.spinics.net/lists/linux-media/msg279805.html

If devs will turn out to be NOT amenable then it would be possible to modify coreboot to introduce a toggable CFR menu option that would allow to force setting board name as DMI Product Name.

In UEFI setup menu there could be an option called "Board Name As DMI Product Name", which by default would be disabled. Flipping it to enabled would cause DMI Product Name to return "Fizz" instead of variant name.

MrChromebox Utility Script does use this field, along with HWID (stored separately), to determine what kind of device it is dealing with, and for some specific variant fixes.

For compatibility reasons (if you are uncertain) you would flip this option to disabled, and reboot, before using this script. Then flip it back to enabled once you would be done. However, doing so would only be of any use when such a change would be in official ROM, since official script does not support flashing custom ROMs.

Example patch file, to show how such modification could look like:

```
diff --git a/src/mainboard/google/fizz/cfr.c b/src/mainboard/google/fizz/cfr.c
index 7827c059fc..563aeb1297 100644
--- a/src/mainboard/google/fizz/cfr.c
+++ b/src/mainboard/google/fizz/cfr.c
@@ -6,6 +6,31 @@
 #include <intelblocks/cfr.h>
 #include <soc/cfr.h>
 
+static const struct sm_object bn_as_dmi_pn = SM_DECLARE_BOOL({
+	.opt_name	= "bn_as_dmi_pn",
+	.ui_name	= "Board Name As DMI Product Name",
+	.ui_helptext	= "Enabling this will make DMI Product Name use board"
+			  " name instead of variant name. This can fix the"
+			  " issue with Linux CEC driver, which checks for board"
+			  " name in DMI Product Name to enable CEC"
+			  " functionality. WARNING: MrChromebox Utility Script"
+			  " can use this field (along with HWID) to determine"
+			  " what kind of device it is dealing with, and what"
+			  " kind of fixes to apply. If you are unsure about"
+			  " compatibility then before launching this script"
+			  " switch this option to disabled and reboot. Once"
+			  " you are done simply flip it back to enabled.",
+	.default_value	= false,
+});
+
+static struct sm_obj_form smbios = {
+	.ui_name = "SMBIOS",
+	.obj_list = (const struct sm_object *[]) {
+		&bn_as_dmi_pn,
+		NULL
+	},
+};
+
 static struct sm_obj_form system = {
 	.ui_name = "System",
 	.obj_list = (const struct sm_object *[]) {
@@ -46,6 +71,7 @@ static struct sm_obj_form *sm_root[] = {
 	&system,
 	&ec,
 	&power,
+	&smbios,
 	NULL
 };
 
diff --git a/src/mainboard/google/fizz/mainboard.c b/src/mainboard/google/fizz/mainboard.c
index e908b9e635..babd62d75d 100644
--- a/src/mainboard/google/fizz/mainboard.c
+++ b/src/mainboard/google/fizz/mainboard.c
@@ -17,6 +17,7 @@
 #include <static.h>
 #include <stdio.h>
 #include <timer.h>
+#include <option.h>
 
 #include <variant/gpio.h>
 
@@ -196,7 +197,9 @@ const char *fizz_oem_name[] = {
 
 const char *smbios_mainboard_product_name(void)
 {
-	return fizz_oem_name[board_oem_id()];
+	return get_uint_option("bn_as_dmi_pn", 0)
+			? "Fizz"
+			: fizz_oem_name[board_oem_id()];
 }
 
 static void mainboard_init(struct device *dev)
```

To build ROM and flash it you would need to follow:
- https://docs.mrchromebox.tech/docs/support/compiling.html
- https://docs.mrchromebox.tech/docs/firmware/manual-flashing.html

Before `./build-uefi.sh <boardname>`, or simply after `git clone https://github.com/mrchromebox/coreboot.git; cd coreboot` you would need to apply the patch with `git apply --verbose /PATH/TO/file.patch`.

I cannot test it right now, as I am in the middle of working on another set of modifications to coreboot for FIZZ boards, but if you are patient then I will test it, and report back if this works.

MrChromebox, 2025-07-01:
I'm perfectly fine if someone wants to add this themselves, I just have no plans to add it to the official release. Google is horrible inconsistent when it comes to DMI data on ChromeOS devices - many older devices report an empty string for the vendor or device when booting Linux with the stock firmware. While I could add a CFR option to use the firmware name vs the board name, implementing it for all devices would be very laborious and isn't very high on my to-do list