MrChromebox/firmware #764: TAEKO does not boot from eMMC

frafra, 2025-06-08:
Hi,
I have a Lenovo Flex 5i (rev4, according to Chrome OS terminal), with hardware ID TAEKO-QSHQ. I am using the RW_LEGACY firmware (MrChromebox-2503.0 Release, Google_Taeko.14505.704.0 01/11/2024), and installed Fedora 42 Workstation into the 64 GB eMMC memory. The installation succeeded, but I am not able to boot the system from the internal storage using edk2, as it is not able to see the internal boot device.

I am documenting my efforts here: https://gist.github.com/frafra/d50f08db3dedf57843390480aa5e1fd3.

MrChromebox, 2025-06-08:
will need a cbmem boot log to start, to see if the issue is coreboot or edk2

MrChromebox, 2025-06-08:
since this is using RW_LEGACY, and the eMMC obviously works for booting ChromeOS, it has to be an issue with the edk2 eMMC driver. Not sure we can get sufficient debug info out of edk2 though, but I guess we can try. I'll have to give you a special build that logs to cbmem and hope it doesn't overflow the buffer, since there's no other option with the stock firmware

frafra, 2025-06-12:
Is there anything I can help with? I could try to make a special build if there is some documentation.

nullandvoyt, 2025-09-09:
Any luck on this? Is there any progress? Can we help somehow?

MrChromebox, 2025-09-09:
TAEKO seems to use a eMMC to NVMe bridge chip which is not supported by edk2. One would have to port the Linux driver (or another driver) somehow to allow it to be used for booting. That's not something I have the time or energy to do at the moment

Redwing321, 2025-12-03:
I'm sorry I can't test it, because I have since sold said Chromebook

frafra, 2025-12-04:
I can test it during the weekend. Thank you very much! :)

closed 2025-12-07 by frafra