MrChromebox/firmware #767: ASUS Chromebox 3 / CN65 (Teemo): System crashes and restarts when disply goes to sleep mode

grzegorzt, 2025-06-26:
When the display enters sleep mode (either after inactivity or due to a programmatic request), my CN65 fails to wake up on mouse movement or keyboard input after several seconds and restarts instead.

This issue occurs on the latest firmware version, MrChromebox-2503.0 (04/27/2025).

I have tested this behavior in Kubuntu 24.04 (X session), Kubuntu 25.04 (Wayland session), Fedora KDE Plasma Desktop (Wayland), and Ubuntu 25.04 (Wayland). When I disable the option for the display to switch off, everything works as expected.

I haven't noticed anything suspicious in the logs. Occasionally, this issue does not occur, but most of the time, it is easily reproducible. For example, I booted from a live USB with Kubuntu, Ubuntu, and Fedora, and left the live system running for a few minutes. Once the display went to sleep, the system restarted after tens of seconds. The same behavior occurs when Linux is installed on the SSD.

When I manually forced the display to go to sleep using KDE's `kscreen-doctor --dpms off`, the result was the same.

However, when I disabled the display power-saving option, I was able to use the system for an entire day without any crashes.

I also tried changing power management settings in the firmware, but this did not resolve the issue.

MrChromebox, 2025-06-27:
I'm unable to reproduce this on either of my FIZZ boxes, running 2503 firmware, using Windows or Linux, with either S0ix or S3 sleep

grzegorzt, 2025-06-27:
That's interesting, maybe that's related to my box. I'll try to change ssd and memory and will update this issue.

AnthroHeader, 2025-11-05:
Replying here to report I have the same issue, I think.

In my case I use a bunch of CN65s headless as Proxmox hosts (on 2025.03 and 2025.10) and some of them have taken to reboot randomly.
When I connect them to a display to try figuring out what's going on, they work - Proxmox does not get the display to sleep on a timer.

I will try updating to 2025.11 and see it if fixes things.

MrChromebox, 2025-11-05:
I don't keep them around just because it's a few GB every release -- I added the backup/custom flash functions to the script so people could back up themselves and stop asking me =D

This release is working fine on my 2 TEEMO test boxes. booting without a display attached has never been supported, it's an edk2 issue. there's a separate issue open already for this.

AnthroHeader, 2025-11-06:
Thank you for your reply.

I’m surprised to hear that the headless boot issue is a known one, especially since it seems very random and not happening uniformly - and also I’ve been using some of those Chromeboxes for years headless without issue.

Would attaching an HDMI dummy plug allow them to boot reliably?