MrChromebox/firmware #772: Windows 10 bootloader not working (causing a BSOD with DRIVER_PNP_WATCHDOG) on HP Barla

aHumanWithAName-commited-arson, 2025-08-03:
Hello! A while ago (I think some time in 2022 or 2023) an update dropped that removed the Driver menu. However this update also stopped the Windows 10 boot loader from working. It shows the Coreboot logo with the Windows 10 spinner and then after a while (maybe about 10 or 15 minutes) it bluescreens. Coolstar says to inject the drivers into boot.wim and install.wim however once I did this it still didn't work.

MrChromebox, 2025-08-03:
I'm really not going to put in any effort fixing issues running Windows on StoneyRidge devices. They are just crappy devices and won't run Windows well.

aHumanWithAName-commited-arson, 2025-08-05:
Okay. Thanks for the reply.

elizajohnson513-cell, 2025-08-30:
> I'm really not going to put in any effort fixing issues running Windows on StoneyRidge devices. They are just crappy devices and won't run Windows well.

can you at least fix legacy boot

MrChromebox, 2025-08-30:
> can you at least fix legacy boot

that's an issue with the stock firmware, nothing I can do unfortuantely

elizajohnson513-cell, 2025-09-01:
> > can you at least fix legacy boot
> 
> that's an issue with the stock firmware, nothing I can do unfortuantely

what's wrong with the stock firmware. I will fix it then. just gotta lmk what's wrong

aHumanWithAName-commited-arson, 2025-09-13:
> > > can you at least fix legacy boot
> > 
> > 
> > that's an issue with the stock firmware, nothing I can do unfortuantely
> 
> what's wrong with the stock firmware. I will fix it then. just gotta lmk what's wrong

It's an issue with googles not-very-well-made firmware. I might be wrong but I think it's a problem with the 11A G8 EE using the same firmware as the 11A G6 EE.

aHumanWithAName-commited-arson, 2025-09-13:
Also elizajohnson513-cell, Are you trying to run windows 11? I figured out how to run it in Full ROM Firmware. I made a script to select either /dev/mmcblk1 or /dev/mmcblk0 and install windows 11 to it. I figured out that skipping the installer fixes the issue. Although windows 10 still doesn't work with this method. The script can successfully install anything with the win11 bootloader and it should work. It can install win10. But coreboot fails to actually boot it. If you want I can drop the script.