MrChromebox/firmware #780: JOXER-IPBR (Ufs) Missing internal disk in RW_Legaxy boot menu

Lioncat6, 2025-09-10:
RW_Legacy firmware seems to be lacking support for the internal disk for the Joxer IPBR (HP Chromebook plus x360)

Installed RW_Legacy firmware and partitioned with CRAS, successfully installed ubuntu to the 12th and 13th partitions, but only USB devices show up in the Boot menu, with the internal disk missing.

[image]
[image]

Lioncat6, 2025-09-10:
```
Device: /dev/sda
Vendor: WDC     
Model: SDINFDO4-256G   
Firmware: 2218

$ ufs-utils desc -a -p /dev/bsg/ufs-bsg0
```
It's UFS

Lioncat6, 2025-09-10:
From what I can tell from digging into the firmware script, adl_2 is the list of devices that use UFS (Referenced the board name `MARASOV` from this related issue #740)

But then adl_n also has it's own unique firmware because they're based off of the `nissa` board which presumably needs special firmware for its hardware.

So Joxer would need a new device group and its own firmware version then from what I can tell. (I'm just curious about how this works, not trying to get in the way or anything)

Lioncat6, 2025-09-14:
I somehow managed to get a successful build from the _rwl branch but after putting it in a bin and flashing it, it just causes my device to reboot.

MrChromebox, 2025-09-14:
try this one:

[rwl_altfw_adl_n_ufs-mrchromebox_20250914.zip](https://github.com/user-attachments/files/22323046/rwl_altfw_adl_n_ufs-mrchromebox_20250914.zip)

unzip, then flash: `sudo flashrom -p internal -w -i RW_LEGACY:rwl_altfw_adl_n_ufs-mrchromebox_20250914.bin`

Lioncat6, 2025-09-14:
The payload flashes fine, but just results in a ‍ screen. Doesn't cause an instant crash, though.

MrChromebox, 2025-09-14:
I'll have to try and test on an ADL-N board here, though I don't have one with UFS

amcnabb, 2025-10-15:
Thanks for the tip on which one is the debugging port.

For the suspending issues, do you have any sense of whether those are driver bugs or UEFI firmware bugs or something else? I'm temporarily going to copy the /boot and efi partitions to a thumb drive; it will be interesting to see whether I see the suspending issues and whether anything changes when I install the Full UEFI Firmware.

Lioncat6, 2025-10-17:
Firmware issues most likely, as it wasn't quite this bad booting off a flash drive. Basically, upon suspending, the device will make up from sleep after a few seconds due to a random ACPI wakeup call that can't be disabled, and unplugging or plugging in the device has a similar outcome.

Hibernation works fine, however, so I've just switched my default suspend behavior to hibernate for now.

MrChromebox, 2025-10-17:
sigh, hard to keep up when supporting 400+ devices.

amcnabb, 2025-10-25:
I got a SuzyQable and installed the full UEFI firmware. I was able to install Linux to the hard drive, but when I then reboot, it drops to the UEFI shell. When I enter "map" or "map -r", it does not see the UFS disk. I think this means that my full UEFI firmware doesn't include UFS support.

Lioncat6, did you maybe get a one-off version of the full UEFI firmware that includes UFS? Or were there maybe some other changes between the version of the firmware that you installed and the most recent version??

Lioncat6, 2025-10-25:
@amcnabb 
I do seem to be running an older version... I wonder if UFS support was removed (likely unintentionally) in the latest release...

<pre>
 ChromeOS Device Firmware Utility Script [2025-10-20] 
 (c) Mr Chromebox &lt;mrchromebox@gmail.com&gt; 
<font color="#2AA1B3">*********************************************************</font>
<font color="#2AA1B3">**</font><font color="#A2734C">     Device: </font>HP Chromebook x360 14b
<font color="#2AA1B3">**</font><font color="#A2734C"> Board Name: </font>JOXER
<font color="#2AA1B3">**</font><font color="#A2734C">   Platform: </font>Intel AlderLake-N
<font color="#2AA1B3">**</font><font color="#A2734C">    Fw Type: </font>Full ROM / UEFI
<font color="#2AA1B3">**</font><font color="#A2734C">     Fw Ver: </font>MrChromebox-2503.0 (04/27/2025)
<font color="#2AA1B3">**</font>             <font color="#26A269"><b>Update Available (10/13/2025)</b></font>
<font color="#2AA1B3">**</font><font color="#A2734C">      Fw WP: </font>Disabled
<font color="#2AA1B3">*********************************************************</font>
<font color="#2AA1B3">**</font><font color="#26A269"><b> [WP]</b></font><font color="#A2734C"><b> 1)</b></font><font color="#2AA1B3"><b> Install/Update UEFI (Full ROM) Firmware </b></font>
<font color="#2AA1B3">**</font><font color="#26A269"><b> [WP]</b></font><font color="#A2734C"><b> 2)</b></font><font color="#2AA1B3"><b> Restore Stock Firmware </b></font>
<font color="#2AA1B3">**</font><font color="#26A269"><b>     </b></font><font color="#A2734C"><b> 3)</b></font><font color="#2AA1B3"><b> Backup Current Firmware </b></font>
<font color="#2AA1B3">**</font><font color="#26A269"><b> [WP]</b></font><font color="#A2734C"><b> 4)</b></font><font color="#2AA1B3"><b> Flash Custom Firmware </b></font>
<font color="#2AA1B3">**</font><font color="#26A269"><b>     </b></font><font color="#A2734C"><b> C)</b></font><font color="#2AA1B3"><b> Clear UEFI NVRAM </b></font>
<font color="#2AA1B3">*********************************************************</font>
<font color="#A2734C">Select a numeric menu option or</font>
<font color="#C01C28">R</font> to reboot  <font color="#C01C28">P</font> to poweroff  <font color="#C01C28">Q</font> to quit 

</pre>

`[NOTE ]  coreboot-MrChromebox-2503.0-MrChromebox-2503.0 Sun Apr 27 12:52:16 UTC 2025 x86_32 bootblock starting (log level: 7)...`

```
[INFO ]  Disable eMMC GPIO pins.

[INFO ]  fw_config match found: THERMAL=THERMAL_FAN

[INFO ]  Set OEM Variable [0] to 1.

[INFO ]  Set OEM Variable [1] to 0.

[INFO ]  eMMC disabled by fw_config

[INFO ]  fw_config match found: STORAGE=STORAGE_UFS

[DEBUG]  All HSPHY ports disabled, skipping HSPHY loading

[INFO ]  Probing TPM I2C: done! DID_VID 0x504a6666

[INFO ]  Locality already claimed

[DEBUG]  GSC TPM 2.0 (i2c 0:0x50 id 0x504a)

[INFO ]  Firmware version: Ti50/D3C1 RO_A:0.0.58/- RW_B:0.23.190/ti50_common_.0.535-c900b8e5

[INFO ]  Enabling GPIO PM b/c CR50 has long IRQ pulse support

[INFO ]  CBFS: Found 'vbt.bin' @0x68340 size 0x500 in mcache @0x76c0d204
```

MrChromebox, 2025-10-25:
UFS support for JOXER has not changed in the latest release

amcnabb, 2025-11-16:
Hi. I saw in the release notes for 2025.11.01 that there is a related update: "Fixed (hopefully) UFS storage not enumerating under edk2 / unable to boot". So I updated the firmware to the latest version.

I have now confirmed that this works. The system is booting successfully on my joxer laptop. Thanks.