MrChromebox/firmware #786: Installing Linux on Pixelbook (Eve) 2017 - Wifi device not found

googleg, 2025-09-21:
Hello all,

I am trying to install Linux Mint on my Pixelbook (2017). I have installed the latest firmware via the script and then Mint.

The problem is: the wireless interface (an Intel 7260 I believe) is not recognized/showing. I've tried to install the iwlwifi modules but it did not change anything.

It seems the device is not even showing in the list below from lspci, and I think it should...

I've tried Mint, Xubuntu and PopOS! and it's always the same.

Any idea?

`$ sudo lspci
00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers (rev 02)
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 615 (rev 02)
00:04.0 Signal processing controller: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem (rev 02)
00:14.0 USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21)
00:14.2 Signal processing controller: Intel Corporation Sunrise Point-LP Thermal subsystem (rev 21)
00:15.0 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #0 (rev 21)
00:15.1 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #1 (rev 21)
00:15.2 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #2 (rev 21)
00:19.0 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO UART Controller #2 (rev 21)
00:19.2 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #4 (rev 21)
00:1c.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #1 (rev f1)
00:1c.4 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #5 (rev f1)
00:1e.0 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO UART Controller #0 (rev 21)
00:1e.2 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO SPI Controller #0 (rev 21)
00:1e.4 SD Host controller: Intel Corporation Device 9d2b (rev 21)
00:1f.0 ISA bridge: Intel Corporation Device 9d4b (rev 21)
00:1f.2 Memory controller: Intel Corporation Sunrise Point-LP PMC (rev 21)
00:1f.3 Multimedia audio controller: Intel Corporation Sunrise Point-LP HD Audio (rev 21)
00:1f.4 SMBus: Intel Corporation Sunrise Point-LP SMBus (rev 21)
00:1f.5 Non-VGA unclassified device: Intel Corporation Device 9d24 (rev 21)
02:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD SM0032L
`

MrChromebox, 2025-09-22:
There's no reason that the WiFi device shouldn't be present, nothing has changed there in years. Pull a cbmem log and attach it or pastebin it and let's see what coreboot says.

googleg, 2025-09-22:
Hello and thank you so much for your answer. I am attaching the cbmem console log. I can see the "WIFI device" is listed...

[cbmem_console.txt](https://github.com/user-attachments/files/22461192/cbmem_console.txt)

MrChromebox, 2025-09-22:
the WiFi device on EVE is attached to PCIe RP1, which is PCI device 0:1c.0. The cbmem log shows that PCI device 1:00.0 is attached to that root port, and its VID/DID indicate an Intel WiFi module.

Everything looks good on the coreboot side, so we need to look at the dmesg log for whatever distro you're using.

googleg, 2025-09-22:
Glad to hear that things look normal on the coreboot side. I am attaching my dmesg : 

[dmesg.txt](https://github.com/user-attachments/files/22465744/dmesg.txt)

closed 2025-09-22 by googleg

MrChromebox, 2025-10-20:
@avm99963  Looking at the kernel log attached here and the one in the link, the issue appear to be the kernel assigning the PCIe root port (1c.0) a MMIO window above 4GB:
`[    0.262638] pci 0000:00:1c.0: bridge window [mem 0x47f000000-0x47f1fffff 64bit pref]: assigned`

which coreboot has not assigned to it, and the device/driver clearly do not like. Why it works after suspend/resume? if you're using S3, then perhaps the resource map is different and it's getting assigned a BAR below 4GB.

if you can, attach cbmem and dmesg logs from a cold boot and then after s3 sleep/resume so we can compare the difference.

interestingly, I can't reproduce this on my KBL Chromebox with the same WiFi module on kernels 6.16-6.17.4.

avm99963, 2025-10-20:
Here are the log files:

[01_cold_boot_cbmem.txt](https://github.com/user-attachments/files/23000687/01_cold_boot_cbmem.txt)
[01_cold_boot_dmesg.txt](https://github.com/user-attachments/files/23000688/01_cold_boot_dmesg.txt)
[02_after_suspend_and_resume_cbmem.txt](https://github.com/user-attachments/files/23000690/02_after_suspend_and_resume_cbmem.txt)
[02_after_suspend_and_resume_dmesg.txt](https://github.com/user-attachments/files/23000689/02_after_suspend_and_resume_dmesg.txt)

I'm not too familiar with the Kernel so it's hard for me to follow your reasoning, but I'm reading about MMIO and how PCI works in general to try to understand you, even if I cannot contribute any ideas.

Thanks for all the help!

MrChromebox, 2025-10-20:
interesting. nothing in those changes jumps out at me as being wrong per se, but the kernel drivers are so complex these days with tons of workarounds and quirks that it's not surprising that this broke something unintentionally

avm99963, 2025-10-20:
Oh well, thanks for the help anyways! :)

kioohe, 2025-12-10:
> I finally managed to fix it by passing "pcie_aspm=off" to the kernel. Thank you so much for your time, this helped me a lot.

use "pcie_aspm.policy=performance"

better

l1k, 2026-03-12:
This issue was root-caused to an erratum of the Sunrise Point-LP PCH built into the Pixelbook Eve. It cannot reinstate the PCIe clock quickly enough when requested by the wifi card in ASPM  L1.2 state. Intel recommends keeping the source clock enabled at the Root Port during L1.2. A corresponding coreboot fix comprising two commits is pending review here:
https://review.coreboot.org/c/coreboot/+/91650
https://review.coreboot.org/c/coreboot/+/91651